heitzmann/gdstk #4: Installation error

yuelyao, 2020-09-10:
Hi Prof.,
Meet the "Could NOT find BLAS (missing: BLAS_LIBRARIES)" problem, and I googled and tried a lot of means for solution but still failed...My OS is CentOS8, with python 3.6.8, lapack.x86_64 3.8.0-8.el8, CMake 3.18.2
Please kindly let me know how to fix. Regards!

heitzmann, 2020-09-10:
@yuelyao Can you install blas as well and try again?

yuelyao, 2020-09-10:
blas-3.8.0-8.el8.x86_64 is already installed and I can find the .so file under /usr/lib64, but some how cmake cannot.
I tried to add following commands in the CMakeLists.txt
set(BLAS_DIR "/usr/lib64")
find_package(BLAS REQUIRED)
But still not working yet.

heitzmann, 2020-09-10:
You probably need the blas-devel and lapack-devel versions installed as well (or the equivalent in CentOS).
Everything is statically compiled, Gdstk doesn't use shared objects (.so, .dll).

yuelyao, 2020-09-10:
Thanks so much! It is successfully installed.

closed 2020-09-10 by heitzmann