heitzmann/gdstk #5: No release

SF73, 2020-09-23:
Hi,

Looking for how to generate gds from python, I found gdspy (which looks awesome btw).
I came here to start directly with gdstk instead of migrating in the future.
However, no version is available. 
Do you plan to release a Windows version soon or should I use gdspy?

Thanks !

heitzmann, 2020-09-23:
I do plan on releasing an alpha version “soon”, but I'd like to finish writing the initial test suite (#2) first.
I've been busy in the past weeks with my daytime job, but I believe I'll be able to publish a release in October.

heitzmann, 2020-10-03:
First alpha release is available: [Release 0.1.0](https://github.com/heitzmann/gdstk/releases/tag/v0.1.0)

closed 2020-10-03 by heitzmann