heitzmann/gdstk #6: Missing DLL on Windows

grahamnorris, 2020-10-08:
I am getting errors when using gdstk 0.1.0 with python 3.8 and anaconda 2020.02 on Windows 10.

- Create fresh conda environment
-  `pip install https://github.com/heitzmann/gdstk/releases/download/v0.1.0/gdstk-0.1.0-cp38-cp38-win_amd64.whl` 
- Launch python interpreter and run `import gdstk`
- Results in following error:
```
>>> import gdstk
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: DLL load failed while importing gdstk: The specified module could not be found.
```

heitzmann, 2020-10-08:
@grahamnorris I tested on a VM and I get the same error you reported. Could you try  to install OpenBLAS on your environment and see if that fixes the issue?
```
pip install openblas
```
It worked for me. If that is the issue (openblas.dll missing) I'll see if I can force cmake to use a static version, not the dll.

grahamnorris, 2020-10-08:
Yes, that seems to have been the issue. After installing openblas, importing gdstk works.

closed 2020-10-08 by grahamnorris