heitzmann/gdstk #9: Running Cell.add in parallel

isourou, 2020-10-19:
Hi @heitzmann 

Thank you so much for your effort to thrust this project. It is incredibly fast!
We get a 20X increase in speed for a layout with 2.6*10^7 references.
For further speeding up I'd like to ask some technique.

I have lists `cellnamelist`, `xlist`, and `ylist`. Now I'm working with a code:
```
cell_to_add=gdslib.new_cell('Top')
reflist = map(putref, cellnamelist, xlist, ylist)
for ref in reflist:
    cell_to_add.add(ref)

def putref(cellname, x, y)
    return gdstk.Reference(cellname,(x,y))
```
It works but takes a long time, especially on the for loop. I also tried:
```
cell_to_add=gdslib.new_cell('Top')
map(lambda cellname, x, y:addref(cell_to_add, cellname, x, y), cellnamelist, xlist, ylist)

def addref(cell_to_add, cellname, x, y)
    cell_to_add.add(gdstk.Reference(cellname,(x,y)))
```
But it doesn't work (References disappear in the output GDSII layout). If you have any idea for speeding up, please let us know.

Thanks,

heitzmann, 2020-10-19:
@isourou In your second example, you're not using `map` correctly (it returns an iterator, it doesn't actually run anything). In the first, map is iterated in the for loop, that's why it works.

The efficiency is lost in the number of function calls. `gdskt.Cell.add` can receive all cells at once, which decreases the number of calls. I get 25% better efficiency using a single call (although creating the references is still quite time-consuming).

```python
import time
import gdstk

lib = gdstk.Library()

xlist = []
ylist = []
celllist = []
for i in range(1000):
    for j in range(1000):
        xlist.append(3 * j)
        ylist.append(i)
        cell = lib.new_cell(f"{i}-{j}")
        cell.add(*gdstk.text(f"{i}-{j}", 0.6, (0, 0)))
        celllist.append(cell)

## Option 1

cell = lib.new_cell("Top1")

def putref(c, x, y):
    return gdstk.Reference(c, (x, y))

start = time.time()
for ref in map(putref, celllist, xlist, ylist):
    cell.add(ref)
print(time.time() - start)  # 390 μs

## Option 2

cell = lib.new_cell("Top2")

start = time.time()
cell.add(*[gdstk.Reference(c, (x, y)) for (c, x, y) in zip(celllist, xlist, ylist)])
print(time.time() - start)  # 290 μs

lib.write_gds("issue.gds")
```

heitzmann, 2020-10-19:
I've just pushed ea76d157f2d548cc74694465d2253fbaa985d70c, which makes the second option a little more efficient by avoiding the creation of the whole list:
```python
cell.add(gdstk.Reference(c, (x, y)) for (c, x, y) in zip(celllist, xlist, ylist))
```

isourou, 2020-10-20:
Hi @heitzmann, I tried [ea76d15](https://github.com/heitzmann/gdstk/commit/ea76d157f2d548cc74694465d2253fbaa985d70c) and it gets faster from 92.8s to 68.6s! Thank you very much.

closed 2020-10-20 by isourou