heitzmann/gdstk #10: Windows installation/dependency problem

g40, 2020-10-21:
Specifically Windows 10x64 with a vanilla Python 3.7 environment. CMake fails looking for BLAS libraries as below:

Questions arising

```
    CMake Error at R:/apps/CMake/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
      Could NOT find BLAS (missing: BLAS_LIBRARIES)
    Call Stack (most recent call first):
      R:/apps/CMake/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
      R:/apps/CMake/share/cmake-3.16/Modules/FindBLAS.cmake:810 (find_package_handle_standard_args)
      R:/apps/CMake/share/cmake-3.16/Modules/FindLAPACK.cmake:197 (find_package)
      CMakeLists.txt:27 (find_package)
```

heitzmann, 2020-10-21:
@g40 The easiest way to install in windows is to use conda (either [Anaconda](https://www.anaconda.com/) or [Miniconda](https://docs.conda.io/en/latest/miniconda.html)). Create an environment for Gdstk if you want, activate it and install all dependencies with:
```
conda install cmake openblas numpy
```
Then, it should be possible to install from source or from the whl binary in that environment (cmake is only needed if you install from source)

jmb-grip, 2020-10-22:
@heitzmann I don't think openblas is available on Windows (both the conda install page and the GitHub page do not list support for Windows).

jmb-grip, 2020-10-22:
Forgive the ignorance of this question, but does LAPACK need to be installed if using the pre-compiled binary wheel? 

I'm also having trouble with installing gdstk on Windows 10 and here are the steps I followed:
 - Install [CMake](https://cmake.org/download/)
 - Install [Visual studio](https://visualstudio.microsoft.com/) for compiling C++
	 - Make sure to select the option that includes C++ compiling
 - Install [OpenBLAS](http://www.openblas.net/) https://sourceforge.net/projects/openblas/files/
	 - Download the zip folder, extract all, then open cmake-gui and select the source code folder, press configure then generate 
 - Download and install the latest binary wheel from the github releases [page](https://github.com/heitzmann/gdstk/releases)

In an anaconda prompt, make an environment and install the wheel with pip
```bash
conda create --name gds
conda activate gds
conda install cmake numpy matplotlib ipykernel
pip install wheel-file-name.whl
```

Installing this way doesn't give any errors but it when trying to import gdstk within Python I get an `ImportError: module not found`.

I'm guessing this is because I haven't installed LAPACK (it's been a bit challenging to get this to work, as someone new to C++ libraries).

heitzmann, 2020-10-22:
@jmb-grip OpenBlas is definitely available for windows through conda-forge: https://anaconda.org/conda-forge/openblas

The binaries were compiled with conda, so the DLL installed from sourceforge might be incompatible with them. That would be my guess.

I was not able to make everything work through the path you're going, although it should, in principle, be possible. If you don't want to use conda and keep trying with the sourceforge version, maybe try to install from source. You'll need to setup a [development environment for x64](https://docs.microsoft.com/en-us/cpp/build/building-on-the-command-line?view=vs-2019) and go from there.

jmb-grip, 2020-10-23:
Ah, I missed that conda channel, my mistake. I'll try doing everything with conda and see if that works.

jmb-grip, 2020-10-23:
Ok, this seems to work with conda-forge.

It might be worth adding an explicit example to the documentation on how to install on Windows. I think visual studio or some other C compiler may be needed? (not positive on this) and then the following conda set-up:

```bash
conda create --name gds
conda activate gds
conda install cmake numpy
conda install -c conda-forge openblas
```

Then , as usual:
```bash
pip install wheel-file-name.whl
```
or:
```bash
python setup.py install
```

heitzmann, 2020-10-23:
@jmb-grip I believe we'll soon have a proper conda package for Gdstk: https://github.com/conda-forge/staged-recipes/pull/12885

Once that happens, the installation instructions will become much easier, so I don't want to add too much confusion right now.

jmb-grip, 2020-10-26:
That makes sense! A conda recipe will be great.

paulgoulain, 2020-10-26:
Is a standard python installation procedure planned ? I mean, without conda ? Because right now, it's not easy to find openblas dll for python.

heitzmann, 2020-10-26:
Not at this point. I don't use Windows regularly and of all the things I tried, only conda worked. I wasn't able to build from source any other way (but there must be a way to do it, of course), but if anyone is able to do it, patches are more than welcome!

g40, 2020-10-27:
@heitzmann (and others) Thanks for all the suggestions. I ran out of time on this one. Please close unless this thread might be useful to other Windows users.

heitzmann, 2020-10-27:
No problem. I'll keep it open as a reminder until the conda recipe is merged. Then I can update the README with proper instructions and close the issue.

jrmagers, 2020-11-17:
I was able to install gdstk in Windows 10 64-bit (with Anaconda already installed) using the following:

```
conda install -c conda-forge openblas
pip install gdstk-0.1.1-cp38-cp38-win_amd64.whl
```

closed 2020-11-18 by heitzmann

wshanks, 2020-11-18:
For those interested, here is the conda package in conda-forge: https://github.com/conda-forge/gdstk-feedstock