heitzmann/gdstk #13: Any plan to also support oasis format?

wylecho, 2020-10-30:
Hi @heitzmann ,

Glad to know you start this C++ project to enhance the previous Gdspy.

Since you need re-design the whole project,  do you have any plan to also support oasis format handling? In IC industry, oasis is much more common used than gds format now. 

Indeed, I am looking for a commercial friendly gds/oasis handling project, such as gdspy/gdstk with BSL (Klayout follows GPL v3 version). Do you have any idea?

Thanks & regards,

heitzmann, 2020-10-30:
@wylecho Supporting OASIS would require some fundamental changes in the internal data structures currently used by Gdstk, because OASIS allows more features than GDSII. I'm basing this on the public available OASIS specification draft, but I imagine it is close to the final specs. Making those changes would take some time, but I can give it a try.

That said, would you (or your company) be willing to cover the expenses for me to buy the [final specification](https://store-us.semi.org/products/p03800-semi-p39-specification-for-oasis%C2%AE-open-artwork-system-interchange-standard)? This being an open source project I develop on my free time, I don't see how I can justify spending US$300,00 on it.

wylecho, 2020-11-01:
@heitzmann Thanks for your reply.

Yes, supporting OASIS will take much more effort than GDS based on my usage experience (mainly geometry operations such as query/Boolean etc.)on three different GDS/OASIS R/W libraries.

I understand your concern and will try to find a copy of the OASIS final specification.

heitzmann, 2020-11-02:
Sounds good. Right now I've been working on C++ examples for the docs and on a better allocation strategy than simply malloc'ing  everything. After that I can start thinking what would the best strategy be based on the draft.

Do you think it makes sense to also support (a subset of) DXF from the industry point of view? I ask because it seems that would require a lot more changes and it would be better to do it now, when we're in alpha stage. Once we hit a certain point, we should avoid breaking changes...

wylecho, 2020-11-03:
As far as I know, DXF is seldom used in IC industry now. I think we could define which geometry types to be supported, and then design the parsing layer to support R/W of different formats. The [DB module](https://www.klayout.de/doc-qt5/code/module_db.html) of Klayout is a sample.

heitzmann, 2020-11-03:
That's good to know... DXF is quite harder to fully support! At this point, the divergences I found between OASIS and GDSII are not so bad. Here are my thoughts:
- Support for circle: either creating a new element (and all support for it) or as a special case of Polygon. I'm not sure what the best solution is.
- Trapezoids and C-Trapezoids: similar issue to circles.
- Repeats: Add as properties for all elements. In GDSII we can actually generate all the geometry when writing.
- `Reference` will have its API changed. This is the major break I foresee, as it will use repeats to create arrays instead of its internal representation. I still have to make sure the transformations are compatible (GDSII array and OASIS repeats) in the order that the transformations are applied.
- Absolute/relative positioning: ?
- Rational coordinates and unlimited integers would be partially supported. Rational numbers are converted to doubles and integers are limited to 64 bits. Anything else would impact the efficiency of the whole library for features that I assume are seldom used.

heitzmann, 2020-12-13:
Update: work on OASIS support is happening in the oasis branch.

heitzmann, 2021-02-06:
@wylecho I'have just added the necessary Python functions to load and save OASIS files in bcb057df90fd5c21b164b729af7ee5240f92e943, if you want to give them a try.

heitzmann, 2021-02-12:
OASIS support has landed in [release 0.3.0](https://github.com/heitzmann/gdstk/releases/tag/v0.3.0).

closed 2021-02-12 by heitzmann