heitzmann/gdstk #14: Creating multiple hierarchies cells

Baruch10, 2020-11-02:
Hi,
I started using gdstk (very high performance!).
In my code, i am creating 1B random rectangles into 1 cell.
I would like to "copy" the content of the cell (or creating another X rectangles) into another cell and so on..
Actually trying to create multiple hierarchies cells & cells comprising of cells within.  
I am not sure how and where to use the References in my code.. 
Appreciate any help for my code..

==========================
import sys
import random
import gdstk
import time

start_time = time.time() 

num = 1000000000

lib = gdstk.Library(unit=1.0e-9, precision=1.0e-9)
gdscell = lib.new_cell("Cell_ID_1")

for j in range(1, num+1):
    x = random.randint(0, 10400000)
    y = random.randint(0, 11200000)
    q = random.randint(10, 500)
    k = random.randint(10, 500)
    gdscell.add(gdstk.rectangle((x, y), ((x + q), (y + k)), 1, 1))
    
    if j % 1000000 == 0 :
        print("Done...%d million" % int(j/1000000))

end_time = time.time()
print ("Time to generate : %d hours, %d min, %d sec" % \
       (int((end_time - start_time)/3600.0), int(end_time - start_time) / 60, int(end_time - start_time)))

start_time = time.time()

print("Writing GDS file...")
lib.write_gds(r"C:\Users\Baruch\Test_1B.gds")

end_time = time.time()
print ("Time to write gds file : %d hours, %d min, %d sec" % \
       (int((end_time - start_time)/3600.0), int(end_time - start_time) / 60, int(end_time - start_time)))

heitzmann, 2020-11-03:
@Baruch10, do you want to repeat the random rectangles you already created in other places or do you want to create another set of rectangles? I'm not sure I understand your end goal so that I can help you out.

Baruch10, 2020-11-03:
yes repeat.
Is it possible to repeat them into a different cells and the cells will be hierarchy (cell within cell)?

heitzmann, 2020-11-03:
Yes, to repeat the same rectangles exactly in, for example, `gdscell2`, simply add a reference to the first cell:
```python
gdscell2 = lib.new_cell("CELL_ID_2")
gdscell2.add(gdstk.Reference(gdscell, (0, 0))
```
You can add both to a third cell the same way, just keep adding references to previous cells. Note that they will all overlap, unless you change the position or add some other transformation to the reference.

closed 2020-11-08 by Baruch10