heitzmann/gdstk #17: Issue with offsetting polygons from different cells

jrmagers, 2020-11-17:
I'm trying to offset all polygons on layer 25 by -1 um and then add them to a call called "mynewcell". Unfortunately, this doesn't seem to work. Any ideas? Does `gdstk.offset()` require that all polygons be in the same cell?

```
import gdstk

lib = gdstk.read_gds("myfile.gds")

polygons = list()

# make a list of polygons on layer 25 to offset by traversing all cells
for cell in lib.cells:
    for pol in cell.polygons:
        if pol.layer == 25:
            polygons.append(pol)

# offset the polygons      
modified_polygons = gdstk.offset(polygons = polygons, distance = -1.0,
                                 use_union = True, layer = 130)
# add the polygons to a new cell
mynewcell = lib.new_cell("mynewcell")
mynewcell.add(*modified_polygons)

lib.write_gds("myfile_modified.gds")
```

jrmagers, 2020-11-17:
Adding to the above, if I modify the "First GDSII Layout" example to open first.gds, add another rectangle to a new cell, and then write to "second.gds", I only see the second rectangle in KLayout. It seems to me that both rectangles should show up.

```
import gdstk

# The GDSII file is called a library, which contains multiple cells.
lib = gdstk.Library()

# Geometry must be placed in cells.
cell = lib.new_cell("FIRST")

# Create the geometry (a single rectangle) and add it to the cell.
rect = gdstk.rectangle((0, 0), (2, 1))
cell.add(rect)

# Save the library in a file called "first.gds".
lib.write_gds("first.gds")

# open "first.gds" and add another rectangle:

lib2 = gdstk.read_gds("first.gds")
cell2 = lib2.new_cell("SECOND")

rect2 = gdstk.rectangle((4, 2), (8, 16),layer=1)
cell2.add(rect2)

lib2.write_gds("second.gds")
```

heitzmann, 2020-11-17:
@jrmagers I believe both problems you're describing come from the same issue. Your files will have more than one “top” cell (i.e. a cell that is not used as a reference elsewhere) so your visualizations software has to choose one of them to show. You are probably looking at cell FIRST in this last example, but you want to look at SECOND. What software are you using to visualize the GDSII files? In KLayout, for example, on the left sidebar you'll find a list of you top cells and they dependence tree. Right click on any of them an choose to “show as net top” to visualize that cell.

jrmagers, 2020-11-18:
@heitzmann Thank you! This is definitely the case in the previous example. Adding the second rectangle (rect2) to one of the top cells fixed the problem. I will revisit the code in my initial post and then close this issue if it works.

Here's the code that works for posterity:

```
import gdstk

# The GDSII file is called a library, which contains multiple cells.
lib = gdstk.Library()

# Geometry must be placed in cells.
cell = lib.new_cell("FIRST")

# Create the geometry (a single rectangle) and add it to the cell.
rect = gdstk.rectangle((0, 0), (2, 1))
cell.add(rect)

# Save the library in a file called "first.gds".
lib.write_gds("first.gds")

# open "first.gds" and add another rectangle:

lib2 = gdstk.read_gds("first.gds")

rect2 = gdstk.rectangle((4, 2), (8, 16),layer=1)

# add rect2 to the top level:
lib2.top_level()[0].add(rect2)

lib2.write_gds("second.gds")
```

closed 2020-11-18 by jrmagers