heitzmann/gdstk #18: offset all polygons that exist on the same layer

jrmagers, 2020-11-18:
Could someone recommend a strategy to "offset" (using `gdstk.offset`) all polygons that exist on the same layer but are in different references? It would convenient to have all of these offset polygons in a single new cell. The offset polygons should retain their position.

heitzmann, 2020-11-18:
It would be easier to understand your goal with an example. Can you provide a simple input and the desired output?

Do you want to retain the original polygons? Does your geometry contain multiple hierarchies (multiple reference levels)? Are there cells you don't want to change, or cells you use in more than one reference and want to change only in 1 place?

The simplest case could be addressed by something like this:

```python
for ref in main_cell.references:
    cell = ref.cell
    old_polys = [p for p in cell.polygons if p.layer == some_layer]
    new_polys = gdstk.offset(old_polys, distance)
    cell.remove(*old_polys)
    cell.add(*new_polys)
```

jrmagers, 2020-11-18:
I attached an example GDS file that I'm working on. Here is what I want to do:

1. make a copy of all polygons on layer 25.
2. put the copies on a separate layer (layer 225); and I'm guessing they would be in a different cell.
3. retain the original polygons.
4. boolean add the copied polygons on layer 225 such that there aren't overlapping polygons.
5. offset all polygons on layer 225 by +2 um.

I believe the part that I can't figure out is step 2.  I can't seem to get copies of the polygons an another layer (in a different cell) and retain their position and orientation.

Thank you for your help!

[project.zip](https://github.com/heitzmann/gdstk/files/5563353/project.zip)

heitzmann, 2020-11-19:
Got it. The problem is that you're going about your problem in the wrong order, making it more difficult than it is. You don't need to change layers or add to a new cell before offsetting, since the operation will create new polygons in any layer you want afterwards.

The trickiest part is actually collecting all polygons from the hierarchy. The easiest way to do so is to copy the main cell and flatten that copy. If the layout is large, it makes sense to filter out all elements not in layer 25 before flattening, but in your case it doesn't make much difference. It would only make the code more complex for little gain.

Once you collect all elements (polygons and paths) from layer 25, it's just a matter of calling [`gdstk.offset`](https://heitzmann.github.io/gdstk/geometry/gdstk.offset.html).
The resulting geometry can be add directly to the main cell or a new one, that can be referenced into main.

Here's the example:
```python
import gdstk

lib = gdstk.read_gds("/home/lucas/Downloads/project.gds")

cell_dict = {c.name: c for c in lib.cells}
main_cell = cell_dict["main"]

main_copy = main_cell.copy("TEMP")
main_copy.flatten()

polygons = [p for p in main_copy.polygons if p.layer == 25]
paths = [p for p in main_copy.paths if p.layers[0] == 25]
new_polygons = gdstk.offset(polygons + paths, 2, use_union=True, layer=225)

main_cell.add(*new_polygons)

lib.write_gds("/home/lucas/Downloads/offset.gds")
```

jrmagers, 2020-11-19:
This is great! Thank you.  

Is there a method to "union" geometry (of a list of polygons)? I didn't see one available in the API, but clearly the `gdstk.offset()` is able to do it. I was able to do it with a hack:
```
cell.remove(*polygons)
polygons = gdstk.offset(polygons, 0.0, use_union=True, layer=18)
cell.add(*polygons)
```

heitzmann, 2020-11-20:
Yes, you can use [`gdstk.boolean`](https://heitzmann.github.io/gdstk/geometry/gdstk.boolean.html):
```python
cell.remove(*polygons)
# Boolean needs 2 operands, so the polygons list is split in 2.
polygons = gdstk.boolean(polygons[0], polygons[1:], "or", layer=18)
cell.add(*polygons)
```

closed 2020-11-21 by jrmagers