heitzmann/gdstk #19: gdstk import error in Jupyter (Anaconda)

SatyajitKautkar, 2020-11-19:
Hi,

I install openblas and gdstk in Anaconda as mentioned in documentation on an offline machine - [Installation via Conda](https://github.com/heitzmann/gdstk#instalation-via-conda)

I installed gdstk using prebuild whl package for windows (as the machine is not connected to internet)

I can see gdstk in packages installed in Anaconda. I was able to successfully installed it.
Below are the logs during gdstk whl package installation

```
(base) C:\Users\Administrator\Downloads\GDSTK>pip install gdstk-0.1.1-cp38-cp38-win_amd64.whl
Processing c:\users\administrator\downloads\gdstk\gdstk-0.1.1-cp38-cp38-win_amd64.whl
Requirement already satisfied: numpy in c:\programdata\anaconda3\lib\site-packages (from gdstk==0.1.1) (1.18.5)
Installing collected packages: gdstk
Successfully installed gdstk-0.1.1
```

But it is giving an error while importing the library

```
ImportError                               Traceback (most recent call last)
<ipython-input-1-228b9c206d31> in <module>
----> 1 import gdstk

ImportError: DLL load failed while importing gdstk: The specified module could not be found.
```

Thank you in advance!

heitzmann, 2020-11-19:
Hi @SatyajitKautkar,

You also need openblas installed. That's the DLL you're missing. We now have a proper conda recipe, so the whl package is deprecated (exactly because of the DLL issue).

SatyajitKautkar, 2020-11-19:
Is it possible to execute that conda recipe for an offline machine?

heitzmann, 2020-11-19:
I don't really know. You'd have to check the conda documentation for offline installation.

SatyajitKautkar, 2020-11-19:
Actually, before installing gdstk, I installed openblas v0.3.12. I can see that in packages installed in Anaconda.

SatyajitKautkar, 2020-11-19:
> I don't really know. You'd have to check the conda documentation for offline installation.

Thanks, I will try this.

closed 2021-02-08 by heitzmann