heitzmann/gdstk #20: Unable to find gdstk in poetry during runtime

rkrishnasanka, 2020-11-19:
When I install gdstk using poetry on my machine (Mac OS pyenv 3.8.0), the python environment is unable to find the gdstk during runtime. It doesn't give me any errors or anything.

```
import gdstk

lib = gdstk.Library()

cell = lib.new_cell("FIRST")

rect = gdstk.rectangle((0,0), (2,1))

cell.add(rect)

lib.write_gds("first.gds")

cell.write_svg("First.svg")
```

rkrishnasanka, 2020-11-19:
I ended up making some progress. It seems like, for whatever reason, poetry is refusing to resolve the library name. I compiled the library on my own and installed it into the system pip (works fine). However, when I try to install into the poetry variant, it basically dies on me.

The only alternative I can think of is decoupling the cmake and the cython part of the project so that I can just create a wheel. Thoughts ?

heitzmann, 2020-11-20:
@rkrishnasanka I've never used poetry, but it is really strange to me that it doesn't produce any errors. What is the output from your script if you add, for example `print(lib)` before exiting?

If I understand correctly, you said it works when you run outside the poetry environment. Do you see the output files?

rkrishnasanka, 2020-11-25:
Nothing, for whatever reason, as soon as it sees the seeing  `import gdstk` it just breaks. Funnily enough, it works fine with pip, so I'm just trying to figure out how to make this work. I'm not yet sure how to compile a `.whl` file for your project (cython + c++). I'll keep you posted if I make progress. I'm planning to use this long term in some of my projects so I'm incentivized to get this to work.

rkrishnasanka, 2020-11-26:
It seems like I was not checking it correctly, it seems to work without a hitch. I'm not sure why this wasn't working the last time.

closed 2020-11-26 by rkrishnasanka