heitzmann/gdstk #21: no install target

francesco-st, 2020-11-27:
I was creating a [lhelper](https://github.com/franko/lhelper.git) recipe for gdstk but I noticed that there is no "install" target defined in the CMake configuration.

I suggest to add an install target if possible. The idea is that install takes care of installing what needed and make the library discoverable. This usually means installing CMake-related files and provide a pkg-config file.

heitzmann, 2020-11-27:
@francesco-st Thanks for the suggestion! I believe commit ce88492235bddbd26f18ec2baf743fb76c0934eb does what you're looking for. Let me know if you have any trouble with it.

francesco-st, 2020-12-04:
Now it works, thank you very much, I appreciate.

closed 2020-12-04 by francesco-st