heitzmann/gdstk #23: Efficient flattening by layer

gambino19, 2021-01-26:
Hello,
First of all, I have really enjoyed using this project, so thank you for the development!
I have a question regarding more efficiently flattening only by layers. Looking at previous issues (specifically #18), you had flattened the hierarchy and then iterated through all polygons and filtered by layer. In my case, I am working on files in the magnitude of E+06 polygons before flattening, so doing a flatten then filter is going to be very time expensive. My current approach is iterating through all library cells + respective polygons, and removing polygons from cell if they don't match the layer(s)
```python
lib = gdstk.read_gds('/path/to/gds') # GDS file size in range of 100MB to 10Gb
for cell in lib.cells:
   polys = [poly for poly in cell.polygons if poly.layer not in [2, 7, 10]] # Arbitrary layers
   cell.remove(*polys)
```
Most of the time in this iteration is spent on the ```cell.remove``` function, as appending all layers by itself would return after a couple of seconds. I use this opposed to creating a copy then ```cell.add``` to maintain the dependencies/references/paths/labels.
Do you see a better way of performing such a method? All advice and feedback is much appreciated. Thank you

heitzmann, 2021-01-27:
@gambino19 Unfortunately removing an element is linear in the number of elements in the cell and the number of elements to be removed, that's why it tales a while for large layouts.

Flattening by layer seems to me like combining 2 functions: filtering by layer and flattening. I'd prefer to keep them independent because we don't always want to flatten a layout just for filtering.

That said, maybe we could introduce a filtering function to the Cell class, as it seems to be a common operation.
It would do something like this (but implemented in C for efficiency):

```python
def filter(self, layers=None, datatypes=None, texttypes=None, condition="and", invert=False):
    """Remove elements with specific layer and data type (text type).

    Args:
      ...
    """
    if condition == "and":
        if invert:
            test = lambda layer, datatype: not (layer in layers and datatype in datatypes)
        else:
            test = lambda layer, datatype: layer in layers and datatype in datatypes
    else:
        if invert:
            test = lambda layer, datatype: not (layer in layers or datatype in datatypes)
        else:
            test = lambda layer, datatype: layer in layers or datatype in datatypes
    for poly in self.polygons:
        if test(poly.layer, poly.datatype):
            cell.remove(poly)
    # ... and similarly for paths and labels.
```
I'm not including references in the filter operation because it is easy enough to loop over all cells in a library or over the dependencies of any given cell.

What do you think?

heitzmann, 2021-01-27:
@gambino19 I forgot: as for your immediate problem, maybe you can try to write your program in C++ instead of Python for now. The C++ example in [this tutorial](https://heitzmann.github.io/gdstk/how-tos.html#geometry-filtering) should be a good start.

gambino19, 2021-01-27:
Agreed about the separation of flatten and filter by layer. But yes, I think that would be a good operation to include into the Cell class, thanks for the warm reception to the idea.
To your other point, I compile with Cython then import as a module into other scripts needing this functionality, but I can try implementation through C++ and see the difference in run time.

closed 2021-02-15 by heitzmann