heitzmann/gdstk #24: use gdstk.Polygon.fracture() to split one rectangle into many adajacent rectangles

miaomiaodepiao, 2021-02-10:
`
#python
import gdstk
points = [1 + 1j, 2 + 1j, 3 + 1j, 4 + 1j, 4 + 0j, 3 + 0j, 2 + 0j, 1 + 0j]
t = gdstk.Library()
cell = t.new_cell("random")
polygon = gdstk.Polygon(points)
poloys = polygon.fracture(max_points=4)
cell.add(*poloys)
t.write_gds("qq_show_points.gds")
`
when the parameter of max_points is 4,the result is None.But I make max_points=5 or 6,the number of points is 4,but the rectangle shrinks.

heitzmann, 2021-02-10:
@miaomiaodepiao You're using the wrong tool for the job. [`Polygon.fracture`](https://heitzmann.github.io/gdstk/geometry/gdstk.Polygon.html#gdstk.Polygon.fracture) will not work for `max_points < 5`, so an empty list is returned.

If you want to split a polygon, use [slice](https://heitzmann.github.io/gdstk/geometry/gdstk.slice.html):

```python
import gdstk

t = gdstk.Library()
cell = t.new_cell("random")

points = [1 + 1j, 4 + 1j, 4 + 0j, 1 + 0j]
polygon = gdstk.Polygon(points)

slices = gdstk.slice(polygon, [2, 3], "x")
for s in slices:
    cell.add(*s)

# include the original in a different layer just to check
cell.add(polygon)
polygon.layer = 1

t.write_gds("qq_show_points.gds")
```

miaomiaodepiao, 2021-02-10:
@heitzmann oh, thank you for your reply.I do misunderstood Polygon.fracture's use.

heitzmann, 2021-02-10:
No problem!

closed 2021-02-10 by heitzmann