heitzmann/gdstk #26: Cell elements ordering

jente-vds, 2021-02-25:
It might be a nice idea to provide a way to order the elements in a cell. Some e-beam systems write the elements in the order in which they are added to the cell, which can cause settling time issues for larger beam movements. A way to be able to arrange the order of the elements in the cell would be handy to counteract this.

heitzmann, 2021-02-25:
@jente-vds Thanks for the input. If I understand correctly, this should not be hard to do at all from the user side:

```python
def polygon_key(polygon):
    # Sort polygons by bounding box
    bb = polygon.bounding_box()
    return bb[0]

polygons = cell.polygons()

# Removing in reverse order is actually more efficient, but the best approach is
# probably to replace the cell with a new one and copy labels, paths and references
cell.remove(*polygons)

polygons.sort(key=polygon_key)
cell.add(*polygons)
```

Is that what you were looking for or am I missing something?

jente-vds, 2021-02-25:
No, you are absolutely right that this would work, thank you. I merely thought this might be handy to implement. Thank you for the quick reply.

closed 2021-02-25 by jente-vds