heitzmann/gdstk #28: Make Precision + Unit attributes public in Python Wrapper

gambino19, 2021-02-26:
In Documentation for gdstk.Library, precision + unit are listed as attributes for Library object. In using python wrapper, it is not possible to access those attributes
```python
lib = gdstk.Library(precision=1e-10)
dir(lib)
>>> [... dundermethods...,
 'add',
 'cells',
 'delete_property',
 'get_property',
 'name',
 'new_cell',
 'properties',
 'remove',
 'set_property',
 'top_level',
 'write_gds',
 'write_oas'] #missing precision + unit
```
For the cases```gdstk.read_oas``` and ```gdstk.read_gds```, it would be nice to access those attributes from constructed library to compare against results from methods like ```gdstk.oas_precision```

heitzmann, 2021-02-26:
This is already available in b8b36442f281ca4da6bd4b85a4ca1b20d89ca092, it's just not yet released.

gambino19, 2021-02-26:
I see, thanks. Will close and update when released.

closed 2021-02-26 by gambino19