heitzmann/gdstk #29: Inconsistent reference bounding box in DRC-generated Oasis file

gambino19, 2021-02-26:
I am running into an issue in which Oasis files generated from the result of Calibre SVRF DRCs (precision 10x old Oasis) are being inconsistently read upon every instance
```python
top1 = gdstk.Library("cell_6022.oas").top_level()[0] #'cell_6022.oas' is new oas generated from DRC 
top2 = gdstk.Library("cell_6022.oas").top_level()[0]
assert top1.name == top2.name
print(top1.bounding_box() == top2.bounding_box())
>>> False
#top1 bbox: ((518.955, 4030.955), (4337.929, 7879.785)
#top2 bbox: ((518.955, 4030.955), (1167.1, 4127)
#Actual bbox: ((518.955, 4030.955), (584.005, 4115.005)
```
Sometimes I will get a Library object that matches the actual bounding box value but it is inconsistent. Diving into the cell some more, we can see there are certain ```gdstk.Reference``` that contain different values
```python
for ref1, ref2 in zip(top1.references, top2.references):
   if ref1.bounding_box() != ref2.bounding_box():
      assert ref1.cell.name == ref2.cell.name
      print(ref1.cell.name)
>>> "cell_6012"
```
It tends to be the same sub-cell that is an issue everytime this test is being done. Weirdly enough, when comparing attributes from reference (i.e cell bounding box, repetitions), they all output the same value. Similarly, when flattened, both cells will have identical bounding boxes, which match the actual bounding box.
```python
top1 = top1.flatten()
top2 = top2.flatten()
top1.bounding_box()
>>> ((518.955, 4030.955), (584.005, 4115.005)
top2 = top2.bounding_box()
>>> ((518.955, 4030.955), (584.005, 4115.005)
```
As well, if instead of creating an oasis file from DRC but rather a gds file, the bounding box value is correct
```python
top3 = gdstk.Library("cell_6022.gds").top_level()[0]
assert top3.name == top1.name
top3.bounding_box()
>>> ((518.955, 4030.955), (584.005, 4115.005)
```
I have been using this with a recursive function to find cell hierarchy at a given coordinate. With this issue, a skewed value often causes incorrect cells to be put in like hierarchies. I have seen values go as high as E+300 for some reference bounding box. This issue also only comes up when using SVRF DRC generated Oasis files, as other Oasis files work fine. As well, there are no paths defined in cell, just references + polygons. Am I missing anything important or leaving out any critical definitions/parameters? All thoughts are appreciated. Thank you

heitzmann, 2021-02-27:
Thanks for reporting @gambino19. Any chance you can share one of the files where you see the issue, so I can reproduce here? I'll try to find the problem anyway, but it will definitely be easier with an example.

heitzmann, 2021-03-01:
@gambino19 Indeed I could not reproduce this bug with my layouts, but they are quite simple. Will you be able to share a reproducible example? Also I noted what must have been a typo: instead of `top1 = gdstk.Library("cell_6022.oas")`, I believe you meant `top1 = gdstk.read_oas("cell_6022.oas")`, right?

gambino19, 2021-03-01:
Yes, it was a typo. Just fixed it on the original issue.
Sorry for the delay - I am going through an approval process to get this data released. It will still contain the bug as-is but will be the most simplified means in which it can be reproduced. Will upload ASAP.

gambino19, 2021-03-01:
Yes, it was a typo. Just fixed it on the original issue. Will upload a reproducible example ASAP.

gambino19, 2021-03-08:
Sorry for the delay, here is the Oasis that produces the bug.

heitzmann, 2021-03-08:
@gambino19 I got a notification from this report that you'd attached the problematic file, but I don't see it here anywhere. Have you removed it? If you prefer not to post it, feel free to email it directly to me.

gambino19, 2021-09-22:
Sorry for such a delay, but recently I was able to do a deep dive on this again and was able to identify some problematic repetitions that were causing skewed bounding boxes when calling `top1.bounding_box()` on a unflattened cell (however when flattened, issue would no longer remain). In particular, OasisRepetition Types 0, 5, and 7 (Previous, ExplicitXGrid, and ExplicitYGrid) seemed to be causing all of the problems. While I cannot release the file itself, here is a particular instance of a type 5 repetition issue (example also holds true for type 0 and 7) in a hexdump of the OASIS:

000000377   04 0a 37 53 35 33 35 34  CELLNAME ("A", id=14)
000000385 + 5a 41 4d 35 0e
000000390   1c 17 01                 PROPERTY (id=1)
000000393   08 a5 8f 80 80 80 80 80  value[0]=1957 (type 8)
000000401 + 00

000001957   0d 0e                    CELL (14)
…
000002717   11 ec                    PLACEMENT
000002719   0e                       id=14
000002720   f4 87 8c 0a              x=10584570
000002724   0b                       repetition (type=11)
000002725   05                         n=5
000002726   0a                         grid=10
000002727   c1 13 80 24                xy=6240,23040
000002731   c3 13 80 36                xy=0,57600
000002735   b9 29 80 12                xy=13260,69120
000002739   b3 b4 01 80 12             xy=-44460,80640
000002744   80 4e                      xy=-38220,80640
000002746   b1 18 80 5a                xy=-30420,138240
…
000003871   11 fc                    PLACEMENT
000003873   0e                       id=14
000003874   ec fb 87 0a              x=10551030
000003878   dc d5 a2 1a              y=27546990
000003882   05                       repetition (type=5)
000003883   01                         n=1
000003884   0a                         grid=10
000003885   f0 04                      x=6240
000003887   ac 23                      x=51480

When having an oas with this content, upon calling `top1.bounding_box()` (again, on a unflattened cell), it would return an inconsistent bounding box result. However, if you were to get rid of the type 5 repetition and call `top1.bounding_box()`  on the unflattened cell again, it would return a consistently correct bounding box result. The same applied to the type 7 repetition. Hopefully this provides a little more replicability to the issue.

heitzmann, 2021-09-23:
Thanks @gambino19! Repetitions were not being taken into account in bounding box calculations. It should be fixed now. The OASIS reader has also had a few minor bugs patched, which should also help. (I'm still working on the OASIS writer, though.)

heitzmann, 2021-10-21:
@gambino19 That's really odd. I cannot reproduce the error on my side. When I use your file I see no problem in the bounding box result. This is my output:
```sh
0 ((12.055, 9.747), (15.104999999999999, 9.857))
1 ((12.055, 9.747), (15.104999999999999, 9.857))
2 ((12.055, 9.747), (15.104999999999999, 9.857))
3 ((12.055, 9.747), (15.104999999999999, 9.857))
4 ((12.055, 9.747), (15.104999999999999, 9.857))
5 ((12.055, 9.747), (15.104999999999999, 9.857))
6 ((12.055, 9.747), (15.104999999999999, 9.857))
7 ((12.055, 9.747), (15.104999999999999, 9.857))
...
98 ((12.055, 9.747), (15.104999999999999, 9.857))
99 ((12.055, 9.747), (15.104999999999999, 9.857))
```
And here are the files I've used (both py and cpp versions work the same way), just to make sure: [files.zip](https://github.com/heitzmann/gdstk/files/7388730/files.zip)

heitzmann, 2021-11-05:
Thanks for the info @gambino19. I believe we cant focus on the C++ version, because that's where the root cause should be. Unfortunately, I still can make the error happen on my end, so I've been trying to look for any bugs that might be related to either loading the OASIS file or executing the bounding box calculation.
The test suite has been showing an error in the same bounding box calculation, but it happens sporadically and only on OS X; it could be related.

Could you run a simple test and send me the output so I can check if there is something wrong with the loaded library? Just add a `lib.print(true)` right after the `read_oas` call in `bb_issue29.cpp`?

gambino19, 2021-11-09:
I see, hopefully those issues are related. Here are some of the outputs from `bb_issue29.cpp`  from adding `lib.print(true)` right after `read_oas`
```
...

Library <0x7ffd471061f0> LIB, unit 1e-06, precision 1e-09, 2 cells, 0 raw cells, owner <(nil)>
{0} Cell <0x1be1010> ChildCell, 1 polygons, 0 flexpaths, 0 robustpaths, 0 references, 0 labels, owner <(nil)>
[0] Polygon <0x1be0f50>, count 4, layer 41, datatype 0, properties <(nil)>, owner <(nil)>
Points: Array <0x1be0f58>, count 4/4
(-0.055, -0.055) (0.055, -0.055) (0.055, 0.055) (-0.055, 0.055)
{1} Cell <0x1be1130> ParentCell, 0 polygons, 0 flexpaths, 0 robustpaths, 1 references, 0 labels, owner <(nil)>
[0] Reference <0x1be11d0> to Cell ChildCell <0x1be1010>, at (12.11, 9.802), 0 rad, mag 1, reflection 0, properties <(nil)>, owner <(nil)>
Explicit X repetition <0x1be1208>: 0.21 0.42 0.84 1.05 1.26 1.47 1.68 2.1 2.31 2.52 2.73 2.94
8: (12.055, 9.747) - (15.105, 9.857)

Library <0x7ffd471061f0> LIB, unit 1e-06, precision 1e-09, 2 cells, 0 raw cells, owner <(nil)>
{0} Cell <0x1be1010> ChildCell, 1 polygons, 0 flexpaths, 0 robustpaths, 0 references, 0 labels, owner <(nil)>
[0] Polygon <0x1be1300>, count 4, layer 41, datatype 0, properties <(nil)>, owner <(nil)>
Points: Array <0x1be1308>, count 4/4
(-0.055, -0.055) (0.055, -0.055) (0.055, 0.055) (-0.055, 0.055)
{1} Cell <0x1be1130> ParentCell, 0 polygons, 0 flexpaths, 0 robustpaths, 1 references, 0 labels, owner <(nil)>
[0] Reference <0x1be11d0> to Cell ChildCell <0x1be1010>, at (12.11, 9.802), 0 rad, mag 1, reflection 0, properties <(nil)>, owner <(nil)>
Explicit X repetition <0x1be1208>: 0.21 0.42 0.84 1.05 1.26 1.47 1.68 2.1 2.31 2.52 2.73 2.94
9: (12.055, 9.747) - (15.105, 9.857)

Library <0x7ffd471061f0> LIB, unit 1e-06, precision 1e-09, 2 cells, 0 raw cells, owner <(nil)>
{0} Cell <0x1be1010> ChildCell, 1 polygons, 0 flexpaths, 0 robustpaths, 0 references, 0 labels, owner <(nil)>
[0] Polygon <0x1be1290>, count 4, layer 41, datatype 0, properties <(nil)>, owner <(nil)>
Points: Array <0x1be1298>, count 4/4
(-0.055, -0.055) (0.055, -0.055) (0.055, 0.055) (-0.055, 0.055)
{1} Cell <0x1be10e0> ParentCell, 0 polygons, 0 flexpaths, 0 robustpaths, 1 references, 0 labels, owner <(nil)>
[0] Reference <0x1be1370> to Cell ChildCell <0x1be1010>, at (12.11, 9.802), 0 rad, mag 1, reflection 0, properties <(nil)>, owner <(nil)>
Explicit X repetition <0x1be13a8>: 0.21 0.42 0.84 1.05 1.26 1.47 1.68 2.1 2.31 2.52 2.73 2.94
10: (12.265, 9.747) - (24.275, 9.857) 

Library <0x7ffd471061f0> LIB, unit 1e-06, precision 1e-09, 2 cells, 0 raw cells, owner <(nil)>
{0} Cell <0x1be1010> ChildCell, 1 polygons, 0 flexpaths, 0 robustpaths, 0 references, 0 labels, owner <(nil)>
[0] Polygon <0x1be1180>, count 4, layer 41, datatype 0, properties <(nil)>, owner <(nil)>
Points: Array <0x1be1188>, count 4/4
(-0.055, -0.055) (0.055, -0.055) (0.055, 0.055) (-0.055, 0.055)
{1} Cell <0x1be10e0> ParentCell, 0 polygons, 0 flexpaths, 0 robustpaths, 1 references, 0 labels, owner <(nil)>
[0] Reference <0x1be1300> to Cell ChildCell <0x1be1010>, at (12.11, 9.802), 0 rad, mag 1, reflection 0, properties <(nil)>, owner <(nil)>
Explicit X repetition <0x1be1338>: 0.21 0.42 0.84 1.05 1.26 1.47 1.68 2.1 2.31 2.52 2.73 2.94
11: (12.055, 9.747) - (15.105, 9.857)
 
...
```

heitzmann, 2022-01-07:
I believe I've finally found the bug in the Repetition code. Now it looks obvious… Let me know if this persists.

gambino19, 2022-01-11:
This is it! The `from_hex.oas` and all my other OAS files seem to be working correctly. Thank you for all the support!

closed 2022-01-11 by gambino19

nmz787-intel, 2022-10-27:
was this fix merged into the main branch and releases? I'm getting this when opening a vendor-generated (Virtuoso) OASIS file:
```
[GDSTK] Unknown record type 0x45.
[GDSTK] Unknown record type 0xAE.
[GDSTK] Unknown record type 0xB2.
[GDSTK] Unknown record type 0x5A.
[GDSTK] Unknown record type 0xA5.
[GDSTK] Unknown record type 0x75.
[GDSTK] Unable to read input file. End of file reached unexpectedly.
Traceback (most recent call last):
  File "/usr/pkgs/python3/3.7.4/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/pkgs/python3/3.7.4/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/nmz787-intel/test.py", line 83, in <module>
    lib = gdstk.read_gds(input_file_path)
OSError: Error reading input file.

```