heitzmann/gdstk #31: Question: how do I use the `fontstyle` argument in `cell.write_svg`?

ottumm, 2021-03-09:
I'm exporting a cell as SVG from a GDS file using `cell.write_svg`, but text labels are not included in the resulting SVG. I believe this is because I'm not including the `fontstyle` argument to `write_svg`, but I can't find any usage instructions in [the doc](https://heitzmann.github.io/gdstk/library/gdstk.Cell.html?highlight=write_svg#gdstk.Cell.write_svg).

I'd appreciate any help on this!

heitzmann, 2021-03-09:
@ottumm `fontstyle` is analogous to `style` in `write_svg`: they are dictionaries of css styles (also in dictionary form) indexed by `(layer, type)` (either data type or text type), as in this example:
```python
fontstyle={
    (0, 0): {"stroke": "black", "font-size": "20px"},  # Style for labels in layer 0, text type 0
    (0, 1): {"stroke": "red", "font-size": "16px"},    # Style for labels in layer 0, text type 1
}
```

Nonetheless, the default value (`None`) implies the use of a default styling, not the removal of the labels, so the labels should be in the SVG file you generate. Could you show an example where you see the problem happening? Could it be an scaling factor?

ottumm, 2021-03-10:
Thank you for the quick response!

> Nonetheless, the default value (None) implies the use of a default styling, not the removal of the labels, so the labels should be in the SVG file you generate.

Ah, ok, that's good to know.

> Could you show an example where you see the problem happening?

I'm working with proprietary files, but I'll see if I can create a simpler test case that reproduces. My intuition says that the problem is definitely on my end. I've tried making the `font-size` very large but it doesn't seem to be working.

_However_, the text labels are definitely included in the SVG file itself, so I should be able to iterate on the CSS in that to figure this out.

> Could it be an scaling factor?

It could! Can you say more about that?

ottumm, 2021-03-10:
It was totally the scaling factor! Thanks for putting me onto that!

All of the text elements have `transform="scale(1 -1) translate(...) scale(0.000000)"`.

When I adjust that last `scale(0.000000)` to anything higher then I can see the text.

ottumm, 2021-03-10:
It looks like that's coming from the `cell.magnification` property, is that correct?

heitzmann, 2021-03-10:
It can be, if you have the label in a cell referenced with magnification 0, that magnification will be applied to all of its contents (including labels in the SVG format).

ottumm, 2021-03-10:
> It can be, if you have the label in a cell referenced with magnification 0, that magnification will be applied to all of its contents (including labels in the SVG format).

Ok, that may be what's going on. I'll look into it more, but I think I'm good.

Thanks for all the support on this! I can put a pull request together to update the documentation if that would be helpful.

heitzmann, 2021-03-10:
Thanks, I've already updated the documentation here, I just haven't pushed it yet!

closed 2021-03-10 by heitzmann