heitzmann/gdstk #32: Segmentation fault with FlexPath bend_radius argument

zobristnicholas, 2021-03-13:
Thanks for this great package!

The following code produces a segmentation fault on my machine. I'm using gdstk version 0.3.2 on Mac OS X.

```
import gdstk
path = gdstk.FlexPath((0, 0), width=1, bend_radius=0.1)
path.horizontal(10, relative=True)
path.vertical(10, relative=True)
path.to_polygons()  # anything that causes the path to be written to a file also causes a segfault.
```

The documentation isn't clear whether appending path segments is compatible with the `bend_radius` argument, but it would be useful. It also would align with how the `joins` keyword argument works with appending segments.

In any case, the code probably shouldn't segfault.

heitzmann, 2021-03-13:
@zobristnicholas `bend_radius` works (where applicable) with all path construction primitives. But the idea is that it creates large bends, not rounded joins, which seems to be what you're looking for with such a small radius. If you increase your bend radius above 0.5, it will work as expected. If you want rounded joins, use those, not `bend_radius`.

Now, if you want to configure the radius of the rounded join, that's trickier. You can either use you own rounding function with `joins=my_custom_function` or build the whole path, transform it into polygons and use [`fillet`](https://heitzmann.github.io/gdstk/geometry/gdstk.Polygon.html#gdstk.Polygon.fillet). Personally, I'd use a custom function (the examples in [`FlexPath`](https://heitzmann.github.io/gdstk/geometry/gdstk.FlexPath.html) should help with that).

That said, segfaulting is not the best outcome when the radius is small, so I've included an extra check in the last commit to skip bends when they don't make sense because the radius is less than half the path with (plus any offsets).

zobristnicholas, 2021-03-13:
Thanks for the suggestion. That fixes my issue.

closed 2021-03-19 by heitzmann