heitzmann/gdstk #33: A wheel package might be helpful for Windows user like me

runck, 2021-03-20:
I am trying this wonderful package on Windows 10, but I fail to build it.

1. I tried "pip install gdstk" first, but the error log said "cl.exe" needed. As I use MinGW gcc, so I do not have the huge MSVC things;
2. Then I tried "python setup.py build" with changing the inside command to be with '-G  "MinGW Makefiles"', but another error said "pthread.h" - not found";
3.  I copied a "pthread.h", but another  ".h" shows not found. I thick maybe too many ".h" are needed in this case.

Maybe it is not easy to fit Windows, but I wonder if a released version wiht wheel package will be helpful for us.
Thank you.

heitzmann, 2021-03-20:
@runck Windows users are *strongly* encouraged to use the [conda package](https://github.com/heitzmann/gdstk#conda).

Gdstk needs not only a properly configured [build environment](https://wiki.python.org/moin/WindowsCompilers) for python, but also compatible libraries for numpy, zlib and lapack.

Getting this to work with the officially-supported compiler is already hard enough that we opted for using the conda packages instead of providing a wheel, so I don't think it is advisable to try with mingw.

runck, 2021-03-20:
I get it. Thank you again.

closed 2021-03-20 by runck