heitzmann/gdstk #34: install gdstk on mac through  conda forge

worldsheep, 2021-03-31:
I tried to install gdstk following the instructions in: [https://github.com/conda-forge/gdstk-feedstock](https://github.com/conda-forge/gdstk-feedstock) but I get the following message in my terminal:

```
conda install gdstk
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining conflict for bleach sphinx dask distributed nbconvert _ipyw_jlab_nb_ext_conf pytest jupyterlab astropy numpydoc pip wheel spyd-
```

The examining conflict part is taking forever and I ended up aborting. Has anyone encountered the same problem?

I have just installed anaconda and I am running version: 1.10.0

heitzmann, 2021-03-31:
@worldsheep Could it be an incompatibility due to numpy from different channels? Please take a look at https://github.com/conda-forge/gdstk-feedstock/issues/2 to see if it helps. Otherwise, would you mind opening this issue in the conda-forge recipe repo?

worldsheep, 2021-04-06:
Unfortunately, I couldn't find a solution. I'll open this issue on the conda-forge recipe repo!

heitzmann, 2021-04-06:
Thanks, I'll close this one and follow up over there.

closed 2021-04-06 by heitzmann