heitzmann/gdstk #35: Having issues using gdstk on Windows

KrA40n1, 2021-04-01:
Hello,

I have installed gdstk successfully by updating all conda modules.

However, now when I try to use gdstk I get the following error:

`C:\Users\Nikolay.Kr\anaconda3\python.exe C:/Users/Nikolay.Kr/Documents/Python/XOR/main.py
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "C:/Users/Nikolay.Kr/Documents/Python/XOR/main.py", line 2, in <module>
    import gdstk
ImportError: numpy.core.multiarray failed to import

Process finished with exit code 1`

I think it might be related to numpy version used(at least that is what I found on the Internet). I have upgraded to the newest version but nothing has changed. Maybe if I use the same version it will work well?

Thank you in advance.

heitzmann, 2021-04-01:
@KrA40n1 Maybe it is related to https://github.com/conda-forge/gdstk-feedstock/issues/2 ? Could you check by using the conda-forge channel for numpy?

heitzmann, 2021-04-14:
@KrA40n1 Please also consider the suggestions in this issue: https://github.com/conda-forge/gdstk-feedstock/issues/9

closed 2021-04-14 by heitzmann