heitzmann/gdstk #36: SVG z-index / stacking order by layer #

jmb-grip, 2021-04-01:
I'm trying to display my drawing as an SVG, where all of the paths and polygons are drawn in the order of their layer number (i.e. objects in layer 1 appear on top of objects in layer 0). 

When using cell.write_svg() for cells containing only polygons, it seems like the polygons are drawn in the order they were added to the cell. But if both paths and polygons are included, it seems like paths are drawn last. 

Is there an easy/known way to have both paths and polygons drawn in order of their layer number? 

As a workaround, I can convert all of my paths to polygons and re-order them by layer number. But it seems like it would be generally useful to have all of the objects of a given layer appearing at the same level relative to other layers.

Here's a minimal example of what I mean, where the path appears at the top while the path converted to polygons appears at the bottom:
```
test = gdstk.Cell("test")
test.add(*gdstk.FlexPath([(-5,-5),(8,8)],width=1,layer=0).to_polygons())
test.add(gdstk.FlexPath([(-5,8),(8,-5)],width=1,layer=0))
polys = [
    gdstk.ellipse((1.5, 1.5), 2.5, layer=4),
    gdstk.ellipse((0, 0), 5, layer=1),
    gdstk.ellipse((5, 0), 5, layer=2),
    gdstk.ellipse((2.5, 5), 5, layer=3),
]
test.add(*polys)

test.write_svg('test.svg',
              style={(1, 0): {"fill": "#a8b3bd","opacity":"1"},
                     (2, 0): {"fill": "#58635a","opacity":"1"},
                     (3, 0): {"fill": "#d4b45d","opacity":"1"},
                     (4, 0): {"fill": "#b282e0","opacity":"1"}},
              pad="0%",background="#fff")
```

Thanks!

jmb-grip, 2021-04-01:
Something simple like this is a suitable workaround:

```python
def reorder_by_layer(cell_to_reorder):
    ordered_cell = cell_to_reorder.copy(cell_to_reorder.name)
    path_polys = [poly for path in ordered_cell.paths for poly in path.to_polygons()]
    all_polys = path_polys + ordered_cell.polygons
    poly_layers = [p.layer for p in all_polys]
    all_polys_sorted = [all_polys[i] for i in np.array(poly_layers).argsort()]
    ordered_cell.remove(*ordered_cell.polygons,*ordered_cell.paths)
    ordered_cell.add(*all_polys_sorted)
    return ordered_cell
```

But I think that this issue could be solved by either (a) add an explanation in the docs of the drawing order for write_svg, or (b) adding an option to write_svg for ordering by layer number and/or datatype.

heitzmann, 2021-04-02:
@jmb-grip I believe this could be implemented without much penalty for the current unsorted version, but I'm very busy right now, so it might take a little while before I do it. It'll probably involve some restructuring in the SVG exporting functions that has to be done carefully.

jmb-grip, 2021-04-05:
Sounds good! I would offer to take a crack at it, but I'm not very fluent in C++.

heitzmann, 2021-04-09:
There is an issue with implementing this that I had not foreseen: `FlexPath` and `RobustPath` don't have a single layer (or data type). They can contain multiple parallel paths, each on its own layer, but they are all generated in a single pass. I'm starting to think the easiest solution would be to use an SVG editor to sort the geometry (in fact a simple python script should be able to do it).

I'll keep thinking about the problem, to see if I can find a robust and efficient solution…

heitzmann, 2021-04-09:
Nevermind... All SVG output is as polygons (including paths), so it is possible to create all polygons beforehand, sort them, and then output. It'll require extra memory, but it's not as convoluted as processing paths by parts.

Now it is just a matter of defining the best interface: I'll probably go with the user passing a comparison function that takes 2 polygons.

heitzmann, 2021-04-09:
@jmb-grip Commit 8976960251a4ac0105528aeaa0fb55b5d0dbc74b should implement the SVG stacking ordering. Note that this is only possible *within* a cell, which means that references will still stack on top of each other in the order they appear.

jmb-grip, 2021-04-12:
Great! This seems like a good solution.

closed 2021-04-14 by heitzmann