heitzmann/gdstk #38: Strange treatment of python keyword argument defaults in RobustPath segment and turn

mdaal, 2021-04-10:
Not sure if this is a bug or intentional. . .

If I set  the keyword arguments to their default value of None in a call to segment I get an error.

For example, the code:
```
path = gdstk.RobustPath((0,0),1)
path.segment(np.array([10,1]), width=None, offset=None, relative=True)
```
gives the error:
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
RuntimeError: Argument width must be a number, a 2-tuple, a callable, or a list of those.

The above exception was the direct cause of the following exception:

SystemError                               Traceback (most recent call last)
<ipython-input-8-4d669812fa75> in <module>
----> 1 path.segment(np.array([10,1]), width=None, offset=None, relative=True)

SystemError: <built-in method segment of gdstk.RobustPath object at 0x7fc524a49490> returned a result with an error set
```

To avoid the error, I must have no `keyword = None`  arguments e.g. :
`path.segment(np.array([10,1]), width= 1, offset=1, relative=True)`
or 
`path.segment(np.array([10,1]))`

Is this intentional?   I would have assumed that `path.segment(np.array([10,1]), width=None, offset=None, relative=True)`  would be interpreted as "keep the width and offset the same as previously declared". I like to explicitly write all the `keyword = None` arguments in my coding which is why I notice this.

same error occurs for `path.turn(radius, angle, offset=None, width=None)` and probably other places like `path_gradient=None` in path.parametric

heitzmann, 2021-04-10:
@mdaal That's definitely not intentional and I believe other functions will have similar issues. I'll have to go through all implementations and make sure they correctly accept `None` as a possible value.

heitzmann, 2021-04-10:
I believe the last commit will fix the issue.

closed 2021-04-14 by heitzmann