heitzmann/gdstk #40: Early reversal to segements when using bend_radius in Paths

twildi, 2021-04-14:
Not sure if this is a bug or intentional.

For certain bends parameters the path object reverts to drawing straight segments '_too early_'. 

Let me illustrate:

[image]

Here the top path is drawn using a 80um radius and still has a considerable straight middle section which should allow the 
use of a greater bend radius but setting the bend radius to 81um results in straight segments instead (bottom path).

Here is the code to reproduce this example.

```
import gdstk

lib = gdstk.Library()
cell = gdstk.Cell('TOP')

# The radius in this cell is drawn
path1 = gdstk.FlexPath((0,0), 5, layer=100, bend_radius=80)
path1.horizontal(500)
path1.segment((60, 150), relative=True)
path1.horizontal(500, relative=True)

cell.add(path1)

# No radius is drawn even though there is clearly enough space for it
path2 = gdstk.FlexPath((0,0), 5, layer=100, bend_radius=81)
path2.horizontal(500)
path2.segment((60, 150), relative=True)
path2.horizontal(500, relative=True)

cell.add(path2.translate(50, -50))

lib.add(cell)
lib.write_gds('test.gds')
```

heitzmann, 2021-04-14:
@twildi This happens because the calculation for defining when a bend can be used is based on 90° angles. This has two problems: it is overzealous for obtuse angles (as in your case) and will produce incorrect results for acute angles. Fixing the issue will probably involve rewriting much of the FlexPath construction algorithm, so it might take a little while.

heitzmann, 2021-04-20:
@twildi I believe it is fixed now. Please, let me know if you find any other cases with issues.

closed 2021-05-04 by heitzmann