heitzmann/gdstk #45: Overwrite duplicate cells

TWPnnook, 2021-06-15:
Hello,

I'm having trouble updating cells with new parametric information. I'm using gdstk 0.5, and viewing the output in klayout 0.27.1.

When I run the following: 
```
lib = gdstk.read_gds(infile=file_name, unit=1e-6)
layer = 1

coord_1 = (0, 0)
coord_2 = (20, 10)
poly_rect = gdstk.rectangle(coord_1, coord_2, layer=layer)

cell_rect = lib.new_cell("rectangle")
cell_rect.add(poly_rect)

lib.write_gds(file_name)
```
The output is as expected
[image]
However if I change any elements of the cell, and re-run the code
```
coord_2 = (10, 20)
```
The klayout output is as follows
[image]

Is there anything I'm missing about interacting with gds libraries in gdstk? Apologies for the basic question, and thanks in advance for your help. I used gdspy previously, and the following code runs without issues, albeit it included an "overwrite_duplicate" parameter in the GdsLibrary's new_cell method.

```
lib = gdspy.GdsLibrary(infile=file_name, unit=1e-6)
layer = 1

coord_1 = (0, 0)
coord_2 = (20, 10)
poly_rect = gdspy.Rectangle(coord_1, coord_2, layer=layer)

cell_rect = lib.new_cell("rectangle", True)
cell_rect.add(poly_rect)

lib.write_gds(file_name)
```

heitzmann, 2021-06-15:
@TWPnnook Unfortunately gdstk does not have cell substitution by default. This is for efficiency reasons, but I might implement it in the future (so please leave this issue open).

Meanwhile, you can use:
```python
import gdstk

def substitute_cell(lib, new_cell):
    lib_cells = {cell.name: cell for cell in lib.cells}
    if new_cell.name in lib_cells:
        old_cell = lib_cells[new_cell.name]
        # Remove old cell
        lib.remove(old_cell)
        # Substitute any references
        for cell in lib.cells:
            for reference in cell.references:
                if reference.cell is old_cell:
                    cell.remove(reference)
                    new_reference = gdstk.Reference(
                        new_cell,
                        reference.origin,
                        reference.rotation,
                        reference.magnification,
                        reference.x_reflection,
                    )
                    new_reference.repetition = reference.repetition
                    cell.add(new_reference)
    # Add new cell
    lib.add(new_cell)

file_name = "test.gds"

lib = gdstk.read_gds(infile=file_name, unit=1e-6)
layer = 1

coord_1 = (0, 0)
coord_2 = (20, 10)
poly_rect = gdstk.rectangle(coord_1, coord_2, layer=layer)

cell_rect = gdstk.Cell("rectangle")
cell_rect.add(poly_rect)

substitute_cell(lib, cell_rect)

lib.write_gds(file_name)
```

TWPnnook, 2021-06-16:
Great, thank you so much! The function works like a charm.

closed 2021-06-17 by heitzmann