heitzmann/gdstk #49: Adding cell containing reference to library produces empty gds

jbm91, 2021-07-03:
Hi

I acknowledge that this is quite likely just me not understanding the syntax, but here goes.

Adding references to a cell seems to be the way to use them. When I now add this cell to my library and write to gds, the written gds is empty when viewing it with KLayout. Example:

`import gdstk

lib = gdstk.Library()    

unit_cell = gdstk.Cell("unit cell")
unit_cell.add(gdstk.ellipse(0.0+0j, radius=1,))

lattice = gdstk.Cell("lattice")
lattice.add(gdstk.Reference(unit_cell, columns=5, rows=5,
                        spacing=(2, 2)))

lib.add(lattice)
lib.write_gds("first.gds")
`

This produces a gds file, that when viewed with KLayout shows the cell structure "lattice > (unit cell)", but no holes are produced.
[image]

I apologize if this is just a stupid question.

heitzmann, 2021-07-03:
You have to add the dependency as well, otherwise it will not work:

```python
lib.add(unit_cell)
```

Gdstk does not include dependencies automatically in order to be more efficient. If we added them automatically, we'd aldo have to check for duplicates and that's way more expensive then simply adding a cell.

jbm91, 2021-07-03:
That makes sense, thank you for the swift reply.

closed 2021-07-03 by heitzmann

nmz787-intel, 2022-08-17:
I wonder if this is this the same issue I am experiencing.

I create several libs, each with one cell, and save them to disk.

Then I create a new lib+cell, and add references to the previously generated cells. First I tried by cell-name (which is also the filename I save the prev libs as). Secondly I opened the prev files, grabbed the cell-object, then added the object as a reference.

In each case, when I opened the lib/cell which used the references, the reference names are just the container name itself!

String-based method:
```
import gdstk
import decimal
unit_value = 1e-6 #micrometer is the unit everyone passes data around with
precision = decimal.Decimal(unit_value) / decimal.Decimal(40000)
lib = gdstk.Library(unit=unit_value, precision=precision)
# Geometry must be placed in cells.
cell = lib.new_cell('my_top_level_cell1')
cell.add(gdstk.rectangle((0, -10), (150, 10)))
lib.write_oas('my_top_level_cell1.oasis')

lib = gdstk.Library(unit=unit_value, precision=precision)
# Geometry must be placed in cells.
cell = lib.new_cell('my_reference_based_cell')
cell.add(gdstk.Reference('my_top_level_cell1'))
lib.write_oas('my_reference_based_cell.oasis')

o = gdstk.read_oas('my_reference_based_cell.oasis')
for c in o.cells:
    for r in c.references:
        print(f'in cell ({c.name}) we have ref to ({r.cell.name})')
```

object-based method:
```
import gdstk
import decimal
unit_value = 1e-6 #micrometer is the unit everyone passes data around with
precision = decimal.Decimal(unit_value) / decimal.Decimal(40000)
lib = gdstk.Library(unit=unit_value, precision=precision)
# Geometry must be placed in cells.
cell = lib.new_cell('my_top_level_cell1')
cell.add(gdstk.rectangle((0, -10), (150, 10)))
lib.write_oas('my_top_level_cell1.oasis')

lib = gdstk.Library(unit=unit_value, precision=precision)
# Geometry must be placed in cells.
cell = lib.new_cell('my_reference_based_cell')
old_lib = gdstk.read_oas('my_top_level_cell1.oasis')
old_cell = old_lib.cells[0]
cell.add(gdstk.Reference(old_cell))
lib.write_oas('my_reference_based_cell.oasis')

o = gdstk.read_oas('my_reference_based_cell.oasis')
for c in o.cells:
    for r in c.references:
        print(f'in cell ({c.name}) we have ref to ({r.cell.name})')
```

in both cases, we find this is printed:
`in cell (my_reference_based_cell) we have ref to (my_reference_based_cell)`

and in both cases, klayout fails to open the file with `internal error: ../../../src/db/db/dbLayout.cc:1330 topological_sort () was not true`