heitzmann/gdstk #51: Could provide an interface for Cell or Polygon to transform with user defined 3*3 Matrix ?

johnny3577, 2021-07-22:


heitzmann, 2021-07-22:
@johnny3577, is this the transformation you have in mind:
```
w = m[2, 0] x + m[2, 1] y + m[2, 2]
x' = (m[0, 0] x + m[0, 1] y + m[0, 2]) / w
y' = (m[1, 0] x + m[1, 1] y + m[1, 2]) / w
```
i.e., using homogeneous coordinates in 2D?

johnny3577, 2021-07-23:
Yes. Currently, "Transformations are applied **in the order** of arguments, starting with magnification and translating by origin at the end." If we can provide the function with Matrix argument. User can generate the matrix as user-defined order, and defined any transformations he wants, such as SHEAR which I not found in API list.

heitzmann, 2021-07-23:
Commit 56159d7 adds a general matrix transform argument to the Python API, because it gives no write access to the polygon points. If you're talking about the C++ API, I'd rather have the user implement it on their own. It should be simple enough to write something like:
```cpp
// Assuming m represents a 3×3 matrix in row-major order
void transform(Polygon& polygon, double* m) {
    Vec2* p = polygon.point_array.items;
    for(uint64_t i = 0; i < polygon.point_array.count; i++, p++) {
        double x = p->x;
        double y = p->y;
        double w = 1.0 / (m[6] * x + m[7] * y + m[8]);
        p->x = (m[0] * x + m[1] * y + m[2]) * w;
        p->y = (m[3] * x + m[4] * y + m[5]) * w;
    }
}
```
There isn't much value in adding this to the interface. I assume that anyone who wants to use a matrix transform (either 3×3 or 2×2) probably knows what it means and should be able to write and optimize for their case. The other interfaces (Polygon::translate, Polygon::scale, etc.) can be used by anyone less familiar with geometrical transformations.

closed 2021-09-04 by heitzmann