heitzmann/gdstk #52: Documentation for the definition of boolean operations

ahmednofal, 2021-07-26:
Hello, thank you for your work on this great project, it is awesome. 

I am a little confused by the meaning of the boolean operations. I can find a definition for the not operation on read the docs for [gdstk](https://heitzmann.github.io/gdstk/geometry/gdstk.boolean.html#gdstk-boolean) but not the rest. I can find the rest in readthedocs for [gdspy](https://gdspy.readthedocs.io/en/stable/gettingstarted.html#boolean-operations). However, they are not clear how they are performed. 

For instance if I have polygons of multiples layers in one python array. Lets say they are contained in a squared area (0, 0) (700, 700) === (lower left x, lower left y) (upper right x, upper right y). 

if I have some other polygons from multiple layers in one python array. and they are contained in a squared area (0,0) (900, 900) === (lower left x, lower left y) (upper right x, upper right y).

What would be the result of "AND" ing those two arrays ?

heitzmann, 2021-07-26:
@ahmednofal The operation names are borrowed from the logical operations that are executed. If you think of the polygons as sets, then *and* is the intersection, *or* is the union, *not* is the difference, and *xor* is the symmetric difference between arrays.
The polygons in each array are merged (union) before the operation.

Here's a quick example:
```python
import gdstk

list_1 = [
    gdstk.rectangle((0, 0), (30, 20), layer=1),
    gdstk.rectangle((20, 10), (40, 30), layer=1),
]

list_2 = [
    gdstk.ellipse((0, 20), 10, layer=2),
    gdstk.ellipse((20, 20), 12, inner_radius=8, layer=2),
]

cell = gdstk.Cell("Example")
cell.add(*list_1, *list_2)

delta = 0
for operation in ("or", "and", "not", "xor"):
    delta = delta - 40
    union = gdstk.boolean(list_1, list_2, operation, layer=0)

    for polygon in union:
        polygon.translate(0, delta)

    cell.add(*union)
    cell.add(*gdstk.text(operation, 10, (50, delta + 10), layer=3))

cell.write_svg("example.svg", 5)
```
[image]

ahmednofal, 2021-07-29:
Thank you for this great explanation. So the number of layers does not matter, correct ? Only the shape and its location, so two overlapping polygons with the same shape and location would be merged into one. correct ?

heitzmann, 2021-07-29:
Correct, the layer numbers and data types don't matter at all. If you have two identical polygons and union them, they'll be merged into a single one.

ahmednofal, 2021-07-30:
Thank you. I am closing this issue now. If it would be ok with you. Please include the great explanation above in the readthedocs website next to the other great documentation for this project.

closed 2021-07-30 by ahmednofal