heitzmann/gdstk #53: Segmentation fault (core dumped) on gdstk.slice

ahmednofal, 2021-07-26:
Hello,

Thank you for your work on this great project.
I am getting a segmentation fault on `gdstk.slice` operation. I hooked up gdb as described in [this answer](https://stackoverflow.com/a/10035594). I could view the traceback and it is breaking at this specific line in c++ python/gdstk_module.cpp:1251
`slice(*polygon_array[i], positions, x_axis, 1 / precision, slices);`
I could not retrieve the arguments to the calling `slice_function` because they are optimized out because the project was built with higher level of optimization flag (probably -O3 ).

I tried retrieving the specific arguments to the line 1251 `slice` function call.
I can access all of the arguments and dereference them before executing the function call
`i = 0`
`polygon_array = {capacity = 256, count = 147, items = 0x9315360}`
`positions = {capacity = 0, count = 1, items = 0x7fffffffd458}`
`x_axis = true`
`precision = 0.001`
`slices = (gdstk::Array<gdstk::Polygon*> *) 0x1068a70`
`slices dereferenced = {capacity = 0, count = 0, items = 0x0}`
tried executing this:
`p polygon_array.operator[](0)`
and got this error
`Cannot evaluate function -- may be inlined`

I hope that is enough to pin-point the problem. I tried recording the execution but failed.

heitzmann, 2021-07-26:
@ahmednofal Can you provide a minimal version of the complete code where you see the problem? In C++ it is hard to guess what the problem might be, but it is probably memory-related.

ahmednofal, 2021-07-29:
Hello, thank you for your quick response. I got swamped with other stuff and forgot to get back to this issue.
I can provide the entire file, here it is: 

```
import sys
sys.settrace
import gdstk
import argparse
from pathlib import Path
from utils import *

def fom_density_check(layout_path):
    layout_ut = gdstk.read_gds(layout_path)

    # diff_layer = layer(layout_ut, 65, 20)
    # tap_layer = layer(layout_ut, 65, 44)
    # fomfill_layer = [*layer(layout_ut, 65, 23), *layer(layout_ut, 65, 28)]
    # fommk_layer = layer(layout_ut, 23, 0)
    all_filled_polygons = [*layer(layout_ut, 65, 20), *layer(layout_ut, 65, 44), [*layer(layout_ut, 65, 23), *layer(layout_ut, 65, 28), *layer(layout_ut, 23, 0)]]

    llx, lly, urx, ury = 0,0,2920,3520
    window_size = 700
    step_size = 70
    window_area = 700 * 700
    with open('fom_density_violating_rectangles.txt', 'w') as fom_density_violating_rectangles:
        for x in range(llx, urx+1-window_size, step_size):
            for y in range(lly, ury+1-window_size, step_size):
                failed = False
                density = area(inside(all_filled_polygons, x, y, x+window_size, y+window_size)) / window_area
                if density > 0.57:
                    failed = True
                if density < 0.33:
                    failed = True
                if failed:
                    fom_density_violating_rectangles.write("density is %s for %s %s %s %s"%(density, x, y, x+window_size, y+window_size))

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("-l", "--layout_path", required=True, type=Path, help="layout path")
    args = parser.parse_args()

    layout_path = Path(args.layout_path).absolute()
    fom_density_check(layout_path)

```

heitzmann, 2021-07-29:
@ahmednofal This script is not enough for me to reproduce or even guess the problem. The only gdstk function you're using there is `read_gds`.

ahmednofal, 2021-07-29:
omg sorry so stupid of me I forgot to include the utils.py.
Here it is.
```
import gdstk

def layer(layout, layer_number, datatype):
    container = []
    for acell in layout.cells:
        for apolygon in acell.polygons:
            if (apolygon.layer, apolygon.datatype) == (layer_number, datatype):
                container.append(apolygon)
    return container

def all_polygons_in_layout(layout):
    container = []
    for acell in layout.cells:
        for apolygon in acell.polygons:
            container.append(apolygon)
    return container

def not_empty_layout(layout):
    return len(layout.top_level()) != 0

def write_polygons(list_of_polygons, output_file):
    new_layout = gdstk.Library()
    cell_containing_all = gdstk.Cell('violating_shapes')
    for apolygon in list_of_polygons:
        cell_containing_all.add(apolygon)
    new_layout.add(cell_containing_all)
    new_layout.write_gds(output_file)

def inside(all_polygons, llx, lly, urx, ury):
    remaining = []
    remaining = gdstk.slice(all_polygons, llx, "x")[1]
    remaining = gdstk.slice(all_polygons, lly, "y")[1]
    remaining = gdstk.slice(all_polygons, urx, "x")[0]
    remaining = gdstk.slice(all_polygons, ury, "y")[0]
    return remaining

def outside(all_polygons, llx, lly, urx, ury):
    remaining_left = []
    remaining_down = []
    remaining_right = []
    remaining_up = []
    remaining_left = gdstk.slice(all_polygons, llx, "x")[0]
    remaining_down = gdstk.slice(all_polygons, lly, "y")[0]
    remaining_right = gdstk.slice(all_polygons, urx, "x")[1]
    remaining_up = gdstk.slice(all_polygons, ury, "y")[1]
    remaining = [*remaining_left, *remaining_down, *remaining_right, *remaining_up]
    return remaining

def area(all_polygons):
    total_area = 0
    for apolygon in all_polygons:
        total_area += apolygon.area()

```

heitzmann, 2021-07-29:
I believe the problem could be in this line:
```python
all_filled_polygons = [*layer(layout_ut, 65, 20), *layer(layout_ut, 65, 44), [*layer(layout_ut, 65, 23), *layer(layout_ut, 65, 28), *layer(layout_ut, 23, 0)]]
```
It seems strange that you have a list of polygons inside the outer list. I believe it should be simply:
```python
all_filled_polygons = [*layer(layout_ut, 65, 20), *layer(layout_ut, 65, 44), *layer(layout_ut, 65, 23), *layer(layout_ut, 65, 28), *layer(layout_ut, 23, 0)]
```
The segfault happens because you probably don't have any polygons in the inner list, so gdstk is interpreting it as a list of coordinates and it creates an empty polygon. That should not happen; either we should raise an error or skip the empty list silently.

ahmednofal, 2021-07-29:
Thank you very much for your tip and for the fix I am closing this issue now. I tried your fix in MY code and it worked. There should not have been an inner list (copying and pasting sideeffect, sorry about that). But I guess it is a good catch in terms of memory management in the code base.

closed 2021-07-29 by ahmednofal

heitzmann, 2021-07-29:
It sure is, thanks for reporting!