heitzmann/gdstk #54: Failure to read gds file

ahmednofal, 2021-07-30:
Hello Thank you for the great work on this project. I tried reading this file in gdstk via `gdstk.read_gds('path/to/file')`
and failed. I have been able to read the same gds file in `klayout`. It seems that this feature might need support. Thank you for your work on this.
```
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.
[GDSTK] Record type NODE (0x15) is not supported.
[GDSTK] Record type NODETYPE (0x2A) is not supported.

    layout_ut = gdstk.read_gds(layout_path)
RuntimeError: Unsupported record in file.

```

heitzmann, 2021-07-30:
Nodes are indeed not supported by either gdspy or gdstk. But that should be a warning, not an error, because, except for the nodes, the layout should be fine.

I'll try to work on that asap.

ahmednofal, 2021-07-31:
Thank you for your quick response. I tried reading another gds file and there are many more record types that are unsupported. Is the intention to never support those record types or is the support coming in a later version ?

heitzmann, 2021-07-31:
At this point, there are no plans of supporting nodes as I have not found any other tool that uses them (I believe Klayout [silently ignores](https://github.com/KLayout/klayout/blob/f8e7e3b6e16ac8e90c5d17458f078619854892f9/src/plugins/streamers/gds2/db_plugin/dbGDS2ReaderBase.cc#L358) those records).
If you have a specific use for any of the unsupported records, let me know and we can see if it is possible to support them.

closed 2022-03-18 by heitzmann