heitzmann/gdstk #55: Failure to install from source on centos 7

ahmednofal, 2021-07-31:
Trying to install using `sudo python3.8 setup.py install` and getting this error from cmake
```
cmake -S /home/nofal/eda/gdstk -B /home/nofal/eda/gdstk/build/temp.linux-x86_64-3.8/cmake_build -DCMAKE_INSTALL_PREFIX=/home/nofal/eda/gdstk/build/temp.linux-x86_64-3.8/cmake_build/install -DCMAKE_BUILD_TYPE=Release
CMake Error: The source directory "/home/nofal/eda/gdstk/build/temp.linux-x86_64-3.8/cmake_build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
error: command 'cmake' failed with exit status 1

```

heitzmann, 2021-08-02:
That's strange. Have you modified anything in the source? Can you post the full log?

closed 2022-03-03 by heitzmann

johnny12150, 2022-10-12:
Still have the same issue and below is the Dockerfile that I am using
```
FROM johnny12150/centos7-py:38

WORKDIR /dl-tmp

RUN wget https://cmake.org/files/v3.12/cmake-3.12.3.tar.gz
RUN tar -zxvf cmake-3.12.3.tar.gz
RUN cd cmake-3.12.3 && ./bootstrap && make && make install

RUN python3 -m pip install --upgrade pip && pip3 install gdstk
```

Full error log would be
```
Collecting gdstk
  Downloading gdstk-0.9.0.tar.gz (501 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 501.0/501.0 kB 2.9 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting numpy
  Downloading numpy-1.23.3-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 17.1/17.1 MB 30.8 MB/s eta 0:00:00
Building wheels for collected packages: gdstk
  Building wheel for gdstk (pyproject.toml): started
  Building wheel for gdstk (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
  
  × Building wheel for gdstk (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [7 lines of output]
      running bdist_wheel
      running build
      running build_ext
      cmake -S /tmp/pip-install-w3hnrid6/gdstk_d38c9bab24764af6b18b9bece4f75c06 -B /tmp/pip-install-w3hnrid6/gdstk_d38c9bab24764af6b18b9bece4f75c06/build/temp.linux-x86_64-cpython-38/cmake_build -DCMAKE_INSTALL_PREFIX=/tmp/pip-install-w3hnrid6/gdstk_d38c9bab24764af6b18b9bece4f75c06/build/temp.linux-x86_64-cpython-38/cmake_build/install -DCMAKE_BUILD_TYPE=Release
      CMake Error: The source directory "/tmp/pip-install-w3hnrid6/gdstk_d38c9bab24764af6b18b9bece4f75c06/build/temp.linux-x86_64-cpython-38/cmake_build" does not appear to contain CMakeLists.txt.
      Specify --help for usage, or press the help button on the CMake GUI.
      error: command '/usr/local/bin/cmake' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gdstk
ERROR: Could not build wheels for gdstk, which is required to install pyproject.toml-based projects
```