heitzmann/gdstk #58: Need for a function to return ALL polygons (geometries) in a layer (number, datatype)

ahmednofal, 2021-08-08:
Hello, thank you for your work on this great project. 

For the boolean operations involved in using gdstk we use lists of polygons to operate on the layout. It is very useful to include a function that returns all polygons (which include all shapes, so it should include all geometries, typically this would mean that there is no need for paths). Currently I am trying to implement this in python but I am afraid it uses too much memory in this layer of code. The process is getting killed by linux due to excessive memory consumption.

```
def layer(layout_obj, layer_number, datatype):
    container = []
    for acell in layout_obj.cells:
        acell.flatten()
        for apolygon in acell.polygons:
            if (apolygon.layer, apolygon.datatype) == (layer_number, datatype):
                container.append(apolygon)
    return container
```
Also is it possible to separate the polygon extraction logic from cells? Or is this a GDS requirement? 
There was something similar in [gdspy](https://gdspy.readthedocs.io/en/stable/reference.html?highlight=get_polygons#gdspy.Cell.get_polygons) (but not quite what we need) but it does not look like it exists in gdstk.

heitzmann, 2021-08-09:
@ahmednofal I don't understand why would you need to flatten all cells in a layout. Shouldn't you be iterating only through top level cells? In any case, memory consumption would be roughly the same either in Python or C++, because you have to create all polygons anyway. But maybe there are other ways to save memory if you can explain exactly what you need to do; for example, can you operate on each cell separately? In that case, you should only gather the polygons of a single cell, operate on them, and discard whatever you don't need before going to the next cell.

In any case, the more compact way to gather polygons from a specific layer and data type in a given cell is:
```python
def get_polygons(cell, layer, dtype):
    # Make sure we don't change the original cell and discard the
    # unused flattened geometry when the function returns
    cell_copy = cell.copy(deep_copy=False)
    cell_copy.flatten()
    return [poly for poly in cell_copy.polygons if poly.layer == layer and poly.datatype == dtype]
```

ahmednofal, 2021-08-09:
I need to flatten all the cells because I need to perform a check on all polygons of a certain layer in the entire layout. Otherwise I would have to make the checks work on cells which is not a good idea tbh, because normal DRC rules do not concern themselves with the cells, it is just materials and their (for instance) spacing. So It would make sense to make the code independent of the cells entirely.

I think retrieving a single layer with all its polygons from a layout could be supported in the c++ level via an array of pointers, pointing to the individual polygons. It would be easier to integrate into the current structure of the code (having the polygons be part of the cells).

heitzmann, 2021-08-09:
Even if you need the polygons from all cells in the layout, flattening only the top level will already give you that. If you have cell A referencing cell B, your function will create at least two copies of B's contents: once when you process A, and again when you process B, but maybe that's indeed what you need. DRC rules concern the geometry being fed through a specific process. How that maps to cells or layouts is unspecified: it could be a single cell, or multiple files, depending on the process itself. You are the one who knows the process, so I can only guess. So your function is more or less correct, but you could still save some memory by copying the cell:

```python
def get_polygons(layout, layer, dtype):
    result = []
    for cell in layout.cells:
        cell_copy = cell.copy(deep_copy=False)
        cell_copy.flatten()
        result.extend(poly for poly in cell_copy.polygons if poly.layer == layer and poly.datatype == dtype)
    return result
```

Your suggestion is already what happens in both interfaces. We only create copies when the geometry needs to change, otherwise pointers to the same polygon are passed around. There is a `Cell::get_polygons` in the C++ interface, but that is worse in terms of memory because it always creates new polygons. Accessing `cell.polygons` from python will only give you the pointers as you suggest. The problem is in the flattening method, which HAS to create new polygons because the geometry possibly changes (with new origins, rotations, etc.) and that applies to both Python and C++.

ahmednofal, 2021-08-09:
I have tried both flattening the top_level(and then just extracting the polygons from the top_level cell) and the function you have provided. Both approaches' python processes get killed (all of 6 GB RAM to use get depleted). The odd thing is I can load the entire layer data in klayout using the same machine (same RAM) with the same gds file.

It seems the way to go about working through this issue is to just setup me DRC checks to process individual cells instead of processing the entire layout at one shot.

heitzmann, 2021-08-09:
Well, loading is one thing, flattening all cells multiple times is something altogether different. You're also able to load the whole layout in gdstk. Executing the DRC more locally is definitley a good idea.

ahmednofal, 2021-08-09:
is loading different that what I am doing ? I meant loading as in "retrieving" all polygons of a layer (I am referring to klayout)

heitzmann, 2021-08-09:
I meant “loading the file.” Are you able to flatten all cells in klayout simultaneously?

ahmednofal, 2021-08-10:
I can retrieve a single layer (all shapes in the layout) without the process getting killed(klayout). I think that (in effect) is equivalent to flattening all the cells(gdstk) (which I only do to retrieve all the polygons in the same layer).

heitzmann, 2021-08-10:
If you're not flattening in klayout and it works for your process, the equivalent can be done in gdstk:
```python
def get_polygons(layout, layer, dtype):
    return [
        poly for cell in layout.cells for poly in cell.polygons
        if poly.layer == layer and poly.datatype == dtype
    ]
```
That function will not create any unnecessary copies. It will return all shapes in a specific layer and datatype from all cells in the layout.

That being said, I believe your process might be incorrect, unless you have other rules in place. Consider this simple example:
```python
lib = gdstk.Library("LIB")

cell1 = lib.new_cell("CELL1")
cell1.add(gdstk.rectangle((0, 0), (10, 10), layer=1, datatype=1))

cell2 = lib.new_cell("CELL2")
cell2.add(gdstk.Reference(cell1, columns=2, spacing=(10.01, 0)))

lib.write_gds("test.gds")
```
Now, say you run a DRC to look for small gaps. CELL1 is fine, it only has a rectangle. CELL2 has no geometry, unless you flatten it. So, if you don't, your DRC will be fine, but if you flatten it, then you'll find 2 rectangles with a small gap between them. Isn't it a problem?

My point is DRC should be run on the final geometry you'll be sending for fabrication, after all implied operations have been performed, not on individual cells.

ahmednofal, 2021-08-11:
Yes you are correct. It should be run on flattened versions of the cells to include actual substitution of references. In designs where there are references the `layer` function in klayout produces correct results and returns shapes of references as if they were actual polygons not references. I am not sure how it is implemented but I believe to achieve the same objective in gdstk (which is the superior option in terms of code cleanliness and readability) I would have to loop over all the cells and flattens them. I imagine the reference (internally in gdstk) is just a pointer an actual polygon, and that is why -maybe- there are possible (maybe, huge) gains to have the `layer` function be part of gdstk. Because then you have the ability to iterate over the references  and just retrieve the layer you need and not all layers in the reference (like when you flatten the cell) and end up with huge memory consumption. It might cause a dramatic hit in performance, but we will never know for sure without the actual implementation.

heitzmann, 2021-08-12:
Can you point me to the documentation of this layer function from KLayout? I only use KLayout for visualization.

In any case, I don't think you need to loop over all cells and flatten them. You only need to flatten the top cell you'll be using in you DRC process, because flattening is a recursive operation, i.e., it will automatically flatten all referenced cells.

This operation cannot return only polygon references, because it modifies the polygon geometry (position, scaling, etc.), so a new polygon must be created and transformed accordingly. If you could operate on a single polygon at a time, you could recuse through all references and apply their stacked transforms to a single polygon in the tree, operate on it, release the memory, and move on to the next polygon. But that's not how DRC operates: you need all polygons simultaneously, so you need to hold them all in memory.

heitzmann, 2021-08-12:
Ok, now I see what you mean! So you still only need to flatten a single cell, but automatically discarding anything outside a specific layer/data type. That can be implemented more or less easily. I'll see if I can push a first prototype today or tomorrow.

ahmednofal, 2021-08-12:
Thank you. That is awesome !!!!! :+1: :+1: :100:

ahmednofal, 2021-08-14:
Thank you for the great work on this issue. I tried the new Cell.get_polygons function with the filter layer=<number> and datatype=<number> and the process got killed as well for excessive memory usage. 

This is the code I used:
```
def layer(layout, layer_number, datatype):
    top_cell = layout.top_level()[0]
    return top_cell.get_polygons(layer=layer_number, datatype=datatype)

```

Am I doing it the wrong way ?

heitzmann, 2021-08-14:
No, this should be correct now. It generates copies of all polygons in the layout, but only for the selected layer and data type.

heitzmann, 2021-08-14:
Another solution you could try is to use [`Cell.filter`](https://heitzmann.github.io/gdstk/library/gdstk.Cell.html#gdstk.Cell.filter). You'd load the layout, filter all cells for the layer and data type you want (`cell.filter([layer_number], [datatype], "nand")`), then flatten the top level. That way you first remove all unnecessary geometry, freeing some memory, then create the new geometry through flatten. The downside is that if you have to repeat the operation for another layer and data type, you'll have to reload the file.

ahmednofal, 2021-08-14:
I just tried it. It is very snappy thank you. I will close this issue now and if anything else comes up I will reopen it. Thank you again @heitzmann

closed 2021-08-14 by ahmednofal

heitzmann, 2021-08-14:
Sounds good. Thanks for the feedback and I'm sorry for taking so long to understand the issue!

ahmednofal, 2021-08-14:
Thank you for putting in the effort !!!

heitzmann, 2021-09-04:
@ahmednofal If you're still interested, [`gsdtk.read_gds`](https://heitzmann.github.io/gdstk/library/gdstk.read_gds.html#gdstk.read_gds) now has a `filter` argument that allows loading only specific layer/data types from the GDSII file. That, in addition to the new function [`gdstk.gds_info`](https://heitzmann.github.io/gdstk/library/gdstk.gds_info.html) should be an easier route to working with specific layers in large layouts.