heitzmann/gdstk #59: Need for option to install via pip (pypi module option)

ahmednofal, 2021-08-08:
Hello, @heitzmann thank you for your work on this great project. 

I wanted to ask you if there is any plans to provide a pypi module for this project. It would make the installation a lot easier.

Thank you.

heitzmann, 2021-08-08:
It is [already available](https://pypi.org/project/gdstk)

Except ypu probably won't be ablento install from Windows, because setting up the build environment and installing the proper dependencies on Windows is nearly impossible.

closed 2021-08-08 by heitzmann

pwuertz, 2022-10-13:
> .. setting up the build environment and installing the proper dependencies on Windows is nearly impossible

I think a popular solution for this is to use some sort of CI/CD service like AppVeyor (free for open source projects), use their [AppVeyor Windows images](https://www.appveyor.com/docs/windows-images-software) for building and packaging `gdstk` wheels for all supported python versions, and then publish those on PyPi.

pwuertz, 2022-10-14:
@heitzmann I could try setting up a GitHub Action script for automatically building `gdstk` binary wheels for Windows on Github, if you're interested in supporting `pip install gdstk`?

heitzmann, 2022-10-17:
@pwuertz sure! That would be great! Thanks!

pwuertz, 2022-10-23:
@heitzmann Allright, this seems to do the trick:
https://github.com/pwuertz/gdstk-build/blob/main/.github/workflows/pypi-packages.yml

This is a "wrapper-project" for building the `gdstk` source package and win64 wheels via GitHub Actions, with the option of automatically publishing those to PyPI. Alternatively, the packages produced by this action could be downloaded and published manually.

Depending on a few decisions, I can finish this up as a PR for the `gdstk` repository:

Do you prefer a fully automated publish to PyPI action, or would you like to download and manually publish the packages?

What is the preferred event for triggering this action? Build (& maybe publish) packages when a version-like tag is pushed, or when a release is created on GitHub?

I was trying to think about the pros/cons, but I'm a bit unsure due to the fact that PyPI doesn't allow a replacement or revision of already published packages for a given version.

heitzmann, 2022-10-25:
Thanks for the work @pwuertz! If publishing automatically continues to submit the source package to PyPI together with the wheels, I see no problem in doing it automatically.

The trigger can be version-like tags pushed to master. The Github release I can create afterwards to write the release notes more carefully and include the published binary wheels as well (if it is not to much trouble for you, you can probably create a draft release in the same action with the artifacts already included)

Feel free to send a PR so that your contribution is clear in the repo!

pwuertz, 2022-10-26:
PR created at https://github.com/heitzmann/gdstk/pull/120