heitzmann/gdstk #61: svg Unescaped '<' not allowed in attributes values

Xingtao, 2021-08-31:
Hi, I use c++ library and modified test file from doc/cpp folder,  to generate svgs.
It runs well, but one of the generated svg file warns "Unescaped '<' not allowed in attributes values" when open with Chrome.
The reproducable code as follow (the layer 5 of 'proof_lib.gds' has warning).

```
int main(int argc, char* argv[]) {
    Library lib = read_gds("proof_lib.gds", 0, 1e-2, NULL);
    StyleMap style = {0};
    StyleMap label_style = {0};
    for (uint64_t j = 0; j < lib1.cell_array.count; j++) {
        Cell* cell = lib3.cell_array[j];
        std::cout << cell->name << ", layer " << j << std::endl;
        cell->write_svg(("test_" + std::to_string(j) + ".svg").c_str(), 10, 6,
                        style, label_style, "#222222", 5, true, nullptr);
    }
    ...
}
```

heitzmann, 2021-08-31:
@Xingtao I can't reproduce the warning message. I've checked the SVG and I can't find a '<' character in any attribute either.

Xingtao, 2021-09-01:
the problem line is as follow:
```
<text id="0000021C7F7663B0" class="l0t0" text-anchor="middle" dominant-baseline="text-after-edge" transform="scale(1 -1) translate(-37.48969 -203.34421) rotate(-150) scale(0)LaBeL</text>
```
It seems proof_lib.gds layer 5 has character that erase close qutation and bracket in svg, related to this line 
https://github.com/heitzmann/gdstk/blob/df1706c9cb9b8e29a9585573edf8592f54182c8b/src/label.cpp#L241
which previously was
```
'fputs("\">", out);' 
```

heitzmann, 2021-09-01:
This line will output both the closing quotation mark and the bracket. The change from the last commit was because the order of transforms was incorrect for labels, but it didn't change the quotation or closing bracket. I've also just pushed another fix that correctly sets the magnification for labels when loading GDSII files, but that does not interfere with this issue.

Here's the test I'm running:
```cpp
#include <stdio.h>

#include "gdstk.h"

using namespace gdstk;

int main(int argc, char* argv[]) {
    Library lib = read_gds("tests/proof_lib.gds", 0, 1e-2, NULL);
    StyleMap style = {0};
    StyleMap label_style = {0};
    for (uint64_t j = 0; j < lib.cell_array.count; j++) {
        Cell* cell = lib.cell_array[j];
        char svg_name[32];
        snprintf(svg_name, COUNT(svg_name), "test_%d.svg", j);
        printf("%s: %s\n", cell->name, svg_name);
        cell->write_svg(svg_name, 10, 6, style, label_style, "#222222", 5, true, NULL);
    }
}
```

And here is the resulting svg:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" width="808.99786" height="715.22101" viewBox="-172.61793 -274.75325 808.99786 715.22101">
<defs>
<style type="text/css">
.l5d0 {stroke: #008856; fill: #008856; fill-opacity: 0.5;}
.l4d0 {stroke: #848482; fill: #848482; fill-opacity: 0.5;}
.l1d0 {stroke: #A1CAF1; fill: #A1CAF1; fill-opacity: 0.5;}
.l0d0 {stroke: #F38400; fill: #F38400; fill-opacity: 0.5;}
.l3d0 {stroke: #C2B280; fill: #C2B280; fill-opacity: 0.5;}
.l2d0 {stroke: #BE0032; fill: #BE0032; fill-opacity: 0.5;}
.l0t0 {stroke: #F38400; fill: #F38400; fill-opacity: 0.5;}
</style>
</defs>
<rect x="-172.61793" y="-274.75325" width="808.99786" height="715.22101" fill="#222222" stroke="none"/>
<g id="Cell.copy" transform="scale(1 -1)">
<polygon id="0x55d183815d60" class="l2d0" points="-110,-120 -110,-80 -90,-80 -90,-120"/>
<polygon id="0x55d183815e30" class="l1d0" points="162.5,-75 162.5,-87.5 137.5,-87.5 137.5,-100 112.5,-100 112.5,-75 137.5,-75 137.5,-62.5 162.5,-62.5 162.5,-50 137.5,-50 137.5,-37.5 112.5,-37.5 112.5,-12.5 137.5,-12.5 137.5,-25 162.5,-25 162.5,-37.5 175,-37.5 175,-50 187.5,-50 187.5,-62.5 175,-62.5 175,-75"/>
<polygon id="0x55d183816040" class="l0d0" points="-62.5,-62.5 12.5,-62.5 12.5,-25 25,-25 25,-62.5 62.5,-62.5 62.5,-12.5 75,-12.5 75,-87.5 -62.5,-87.5"/>
<polygon id="0x55d183816170" class="l0d0" points="-62.5,75 -25,75 -25,50 -62.5,50"/>
<polygon id="0x55d183816240" class="l0d0" points="40,126.5165 13.4835,153.03301 26.74175,166.29126 22.32233,170.71068 9.06408,157.45243 -4.19417,170.71068 13.4835,188.38835 9.06408,192.80777 -17.45243,166.29126 31.16117,117.67767"/>
<polygon id="0x55d1838163c0" class="l0d0" points="88.61359,175.1301 75.35534,188.38835 66.5165,179.54951 79.77476,166.29126"/>
<polygon id="0x55d183816490" class="l0d0" points="-38.81806,-216.14501 -57.56806,-248.62096 -73.80603,-239.24596 -76.93103,-244.65862 -60.69306,-254.03362 -70.06806,-270.2716 -91.71869,-257.7716 -94.84369,-263.18426 -62.36774,-281.93426 -27.99274,-222.39501"/>
<polygon id="0x55d1838165c0" class="l0d0" points="47.78448,-266.14501 29.03448,-298.62096 12.79651,-289.24596 9.67151,-294.65862 25.90948,-304.03362 16.53448,-320.2716 -5.11615,-307.7716 -8.24115,-313.18426 24.2348,-331.93426 58.6098,-272.39501"/>
<polygon id="0x55d1838166f0" class="l0d0" points="134.38702,-316.14501 115.63702,-348.62096 99.39905,-339.24596 96.27405,-344.65862 112.51202,-354.03362 103.13702,-370.2716 81.48639,-357.7716 78.36139,-363.18426 110.83734,-381.93426 145.21234,-322.39501"/>
<polygon id="0x55d1838168b0" class="l0d0" points="1.18194,-146.86298 -17.56806,-179.33893 -33.80603,-169.96393 -36.93103,-175.37659 -20.69306,-184.75159 -30.06806,-200.98956 -51.71869,-188.48956 -54.84369,-193.90222 -22.36774,-212.65222 12.00726,-153.11298"/>
<polygon id="0x55d1838169e0" class="l0d0" points="87.78448,-196.86298 69.03448,-229.33893 52.79651,-219.96393 49.67151,-225.37659 65.90948,-234.75159 56.53448,-250.98956 34.88385,-238.48956 31.75885,-243.90222 64.2348,-262.65222 98.6098,-203.11298"/>
<polygon id="0x55d183816b10" class="l0d0" points="174.38702,-246.86298 155.63702,-279.33893 139.39905,-269.96393 136.27405,-275.37659 152.51202,-284.75159 143.13702,-300.98956 121.48639,-288.48956 118.36139,-293.90222 150.83734,-312.65222 185.21234,-253.11298"/>
<polygon id="0x55d183816c40" class="l0d0" points="-98.3573,-181.77001 -107.7323,-198.00798 -96.90699,-204.25798 -87.53199,-188.02001"/>
<polygon id="0x55d183816d10" class="l0d0" points="-11.75476,-231.77001 -21.12976,-248.00798 -10.30445,-254.25798 -0.92945,-238.02001"/>
<polygon id="0x55d183816de0" class="l0d0" points="74.84778,-281.77001 65.47278,-298.00798 76.29809,-304.25798 85.67309,-288.02001"/>
<polygon id="0x55d183816eb0" class="l0d0" points="-58.3573,-112.48798 -67.7323,-128.72595 -56.90699,-134.97595 -47.53199,-118.73798"/>
<polygon id="0x55d183816f80" class="l0d0" points="28.24524,-162.48798 18.87024,-178.72595 29.69555,-184.97595 39.07055,-168.73798"/>
<polygon id="0x55d183817160" class="l0d0" points="114.84778,-212.48798 105.47278,-228.72595 116.29809,-234.97595 125.67309,-218.73798"/>
<polygon id="0x55d183817230" class="l1d0" points="398.4054,118.4799 392.99274,115.3549 399.24274,104.52958 393.83008,101.40458 400.08008,90.57926 410.9054,96.82926 404.6554,107.65458 410.06806,110.77958 403.81806,121.6049 409.23072,124.7299 415.48072,113.90458 420.89337,117.02958 427.14337,106.20426 437.96869,112.45426 [... cut ...]
<text id="0x55d18381e350" class="l0t0" text-anchor="middle" dominant-baseline="text-after-edge" transform="translate(423.88715 -102.65579) rotate(30) scale(1 -1)">LaBeL</text>
<text id="0x55d18381e400" class="l0t0" text-anchor="middle" dominant-baseline="text-after-edge" transform="translate(337.28461 -152.65579) rotate(30) scale(1 -1)">LaBeL</text>
</g>
</svg>
```

All labels are properly formatted, that's why I say I can't reproduce the issue…
I wonder if it could it be an encoding issue. All source files are supposed to be UTF-8.

Xingtao, 2021-09-02:
Thanks,  it is indeed an encoding issue. 
Three trailing dots '...' at the comment line  is not ascii.  when I use VSCode open the file with UTF8, it takes the next line as comment too.
```
    // NOTE: Escape “<”, “>”, and “&” inside the SVG tag.  Here be dragons if the text is not ASCII.
    // The GDSII specification imposes ASCII-only for strings, but who knows…
    fputs(" scale(1 -1)\">", out);
```

closed 2021-09-02 by Xingtao