heitzmann/gdstk #62: No end types in FlexPath python API

jrreinhardt, 2021-09-15:
I might be wrong but I read through the 0.7.0 documentation several times.

It looks like though the API allows the user to create FlexPath with different end types, but there is no field/method to know the end types or related configurations for an existing FlexPath object.

This does cause some trouble when reading an existing GDS to process.

Is there a plan to add this feature?

 

Regards,

ZW

heitzmann, 2021-09-15:
Thanks for the report @jrreinhardt
I've added that and a few other attributes to both `FlexPath` and `RobustPath`, as applicable: `ends`, `joins`, `tolerance`, etc.
Let me know if that solves the issue for you.

jrreinhardt, 2021-09-17:
Thank you for the timely revision. I tested the latest repository and it works for me.

closed 2021-09-17 by jrreinhardt