heitzmann/gdstk #63: Cannot build on Centos 6.9

sgbrickles, 2021-09-24:
Build fails with:

[  2%] Building CXX object CMakeFiles/gdstk.dir/src/clipper_tools.cpp.o
In file included from /apps/gnu/python/src/gdstk-0.7.0/src/clipper_tools.h:16,
                 from /apps/gnu/python/src/gdstk-0.7.0/src/clipper_tools.cpp:8:
/apps/gnu/python/src/gdstk-0.7.0/src/array.h: In member function ‘void gdstk::Array<T>::insert(uint64_t, T)’:
/apps/gnu/python/src/gdstk-0.7.0/src/array.h:87: warning: declaration of ‘index’ shadows a member of 'this'
/apps/gnu/python/src/gdstk-0.7.0/src/array.h: In member function ‘void gdstk::Array<T>::remove_unordered(uint64_t)’:
/apps/gnu/python/src/gdstk-0.7.0/src/array.h:104: warning: declaration of ‘index’ shadows a member of 'this'
/apps/gnu/python/src/gdstk-0.7.0/src/array.h: In member function ‘void gdstk::Array<T>::remove(uint64_t)’:
/apps/gnu/python/src/gdstk-0.7.0/src/array.h:107: warning: declaration of ‘index’ shadows a member of 'this'
In file included from /apps/gnu/python/src/gdstk-0.7.0/src/oasis.h:18,
                 from /apps/gnu/python/src/gdstk-0.7.0/src/polygon.h:18,
                 from /apps/gnu/python/src/gdstk-0.7.0/src/clipper_tools.h:17,
                 from /apps/gnu/python/src/gdstk-0.7.0/src/clipper_tools.cpp:8:
/apps/gnu/python/src/gdstk-0.7.0/src/map.h: In member function ‘gdstk::MapItem<T>* gdstk::Map<T>::next(const gdstk::MapItem<T>*) const’:
/apps/gnu/python/src/gdstk-0.7.0/src/map.h:94: warning: declaration of ‘next’ shadows a member of 'this'
/apps/gnu/python/src/gdstk-0.7.0/src/clipper_tools.cpp: In function ‘void gdstk::link_holes(ClipperLib::PolyNode*, gdstk::ErrorCode&)’:
/apps/gnu/python/src/gdstk-0.7.0/src/clipper_tools.cpp:167: error: no match for ‘operator=’ in ‘p_closest = contour->std::vector<_Tp, _Alloc>::insert [with _InputIterator = __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >, _Tp = ClipperLib::IntPoint, _Alloc = std::allocator<ClipperLib::IntPoint>](p_closest, holes.gdstk::Array<T>::operator[] [with T = gdstk::SortingPath](i)->gdstk::SortingPath::path->std::vector<_Tp, _Alloc>::begin [with _Tp = ClipperLib::IntPoint, _Alloc = std::allocator<ClipperLib::IntPoint>](), hole_min.__gnu_cxx::__normal_iterator<_Iterator, _Container>::operator+ [with _Iterator = ClipperLib::IntPoint*, _Container = std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> >](((const ptrdiff_t&)((const ptrdiff_t*)(&1)))))’
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_iterator.h:669: note: candidates are: __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >& __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >::operator=(const __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >&)
/apps/gnu/python/src/gdstk-0.7.0/src/clipper_tools.cpp:168: error: no match for ‘operator=’ in ‘p_closest = contour->std::vector<_Tp, _Alloc>::insert [with _InputIterator = __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >, _Tp = ClipperLib::IntPoint, _Alloc = std::allocator<ClipperLib::IntPoint>](p_closest, hole_min, holes.gdstk::Array<T>::operator[] [with T = gdstk::SortingPath](i)->gdstk::SortingPath::path->std::vector<_Tp, _Alloc>::end [with _Tp = ClipperLib::IntPoint, _Alloc = std::allocator<ClipperLib::IntPoint>]())’
/usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_iterator.h:669: note: candidates are: __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >& __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >::operator=(const __gnu_cxx::__normal_iterator<ClipperLib::IntPoint*, std::vector<ClipperLib::IntPoint, std::allocator<ClipperLib::IntPoint> > >&)
make[2]: *** [CMakeFiles/gdstk.dir/src/clipper_tools.cpp.o] Error 1
make[1]: *** [CMakeFiles/gdstk.dir/all] Error 2
make: *** [all] Error 2

Any ideas ?  This is with GCC 7.5

Thanks,
Stephen

sgbrickles, 2021-10-11:
Thanks, but we run Centos 6.x systems here which do not have Docker.  We would like to be able to build this on Centos 6.x

ustcjmf, 2022-02-18:
Hello, I have encountered the same problem. Do you have any solution later? Thx in advance.

heitzmann, 2022-02-18:
Do you still get the same error if you change line 137 in src/clipper_tools.cpp from `ClipperLib::Path::iterator p_closest = contour->end();` to `auto p_closest = contour->end();`?

nescirem, 2022-02-23:
> Do you still get the same error if you change line 137 in src/clipper_tools.cpp from `ClipperLib::Path::iterator p_closest = contour->end();` to `auto p_closest = contour->end();`?

It works.

heitzmann, 2022-02-23:
So, it looks like gcc 4.4.7 (which is quite old) has incomplete support for C++ 11. You can patch the package for CentOS by introducing those changes, but I cannot guarantee it would work overall (maybe there are other changes that don't show up as compilation bugs…). On my computer, it seems to compile correctly with gcc 5.5, if you have a possibility of updating.

Either way, supporting such old versions of gcc could end up being more work than I can handle on my free time, so I prefer not to introduce the changes in the main repo.

closed 2022-02-23 by heitzmann