heitzmann/gdstk #64: Does `Cell.get_polygons()` return polygons in a stable order?

jatoben, 2021-09-28:
@heitzmann Thanks for adding the `get_{labels,paths,polygons}` methods recently—they're super handy! From a quick look at the source, it appears that calling `get_polygons()` twice on the same cell will return the polygons in the same order each time; is that correct?

As a bit of background, I have an algorithm which iterates through a cell's polygons and performs some transformation and clipping operations, then adjusts the transformation parameters and iterates through the polygons again (this can repeat many times). I'd like to remember the outcome of the operations in pass one so that I can compare them to pass N. If I know that (for example) the fifth polygon in pass one is the same as the fifth polygon in pass N, that will make tracking the results easy.

For this use case, it's unimportant if the order changes in a future release of gdstk, or if the cell is loaded again from the same GDS file; I'm only interested in the order for a `Cell` object and any copies of that object produced by `Cell.copy()`.

heitzmann, 2021-09-28:
Yes, it should be stable between a cell and its copies (both deep and shallow copies). But I can't guarantee that it will remain so in future releases. At this point, I see no reason to make it “unstable”. It would have to be a significant performance boost to do so, but I can't imagine that being the case.

jatoben, 2021-09-28:
Great—thanks!