heitzmann/gdstk #70: gds_info function do

Xingtao, 2021-11-11:
Hi, 
Before use gds_info function, it has to memset 'LibraryInfo info', otherwise it crash.

This somehow anti-intuitive for C++ programer.
For instance, when 'LibraryInfo info' is a class member, it expects 'LibraryInfo info' has it is own constructor and being initialized.

So, recommand to manully do memset in 'gds_info' function or add a constructor for LibrayInfo.

Thanks for your handy and efficient project.

heitzmann, 2021-11-11:
Hi @Xingtao, the library is written mostly as a C library, but we use a few helpful C++ features. Constructors and destructors, for example, are not used, so that allocation and initialization are kept separate. That means the user is responsible for proper initialization of all the data.

closed 2021-11-12 by Xingtao

wylecho, 2021-11-27:
Hi @heitzmann, I have a quick review of some related structs (Array/Cell/Polygon/Set etc.) and think adding constructors and destructors for them has no bad side effect while could help us get ride of unexpected crash. Initializing each struct before using them is very unusual and most programmers usually forget to do it. 
``
If you agree, I can help adding constructor/destructor for these structs. How do you think?

heitzmann, 2021-11-27:
@wylecho Constructors and destructors do have side effects. For example, when we allocate large arrays (which is a valid use case for this library), they will impact performance. Here's a very simple example: https://godbolt.org/z/qM968xnKo

Initializing each structure might be unusual for C++, but it is definitely usual in C. This library is written in C++ only so that we can use a few limited C++ features (some templating and function/operator overloading in specific cases), nothing else.

wylecho, 2021-11-27:
@heitzmann I agree with you constructors/destructors impact performance. Then do you think it's worthy to add an initialization function for these structs so that we could get ride of writing too many lines of initialization code?
```
    // read in the layout polygons
    gdstk::Set<gdstk::Tag> tag_set;
    // gdstk struct initialization before using
    tag_set.capacity = 0;
    tag_set.count = 0;
    tag_set.items = NULL;
    gdstk::Tag layer_datatype_tag = gdstk::make_tag(in_layout_layer_datatype.x, in_layout_layer_datatype.y);
    tag_set.clear();
    tag_set.add(layer_datatype_tag);
    const double I2C_INTERNEL_UNIT = 1e-9;
    const double I2C_INTERNEL_TOLERENCE = I2C_INTERNEL_UNIT;
    gdstk::ErrorCode error_code = gdstk::ErrorCode::NoError;
    gdstk::Library in_lib = gdstk::read_gds(in_layout_path.c_str(), I2C_INTERNEL_UNIT, I2C_INTERNEL_TOLERENCE, &tag_set, &error_code);
    if (error_code != gdstk::ErrorCode::NoError) {
        spdlog::default_logger()->error("read gds error code: {}", error_code);
        return false;
    }
    gdstk::Array<gdstk::Polygon *> in_layout_polygons;
    // gdstk struct initialization before using
    in_layout_polygons.capacity = 0; 
    in_layout_polygons.count = 0;
    in_layout_polygons.items = NULL;
    for (unsigned i = 0; i < in_lib.cell_array.count; ++i) {
     in_lib.cell_array.items[i]->get_polygons(true, false, -1, true, layer_datatype_tag, in_layout_polygons);
    }
```

heitzmann, 2021-11-27:
Sure, optional initialization functions would not be a problem at all. I believe you're talking about something analogous to ``FlexPath::init`` but for other structures, is that it?

wylecho, 2021-11-27:
Yes, exactly!

heitzmann, 2021-11-27:
Sounds good, then. I have some time to work on it today, so I'll start on the most problematic structs first and go from there.

wylecho, 2021-11-27:
That's great. Thanks so much and take your time.