heitzmann/gdstk #71: polygon vertex order

wylecho, 2021-11-27:
Hi @heitzmann, in my testing, it seems the polygon vertex order is clockwise for hull polygon and conter-clockwise for hole polygon. It means the inside region is always on the right of the vertex order. Can we always take it as true?

heitzmann, 2021-11-27:
Hello, @wylecho! I'm not sure I understand your question. There are no functions in the library that create holes and, internally, all user-defined shapes are considered to be a “hull”. The polygons generated by the library functions should be all counter-clockwise in the usual cases, but it is possible to generate clockwise polygons with “unusual” parameters.

wylecho, 2021-11-27:
In IC layout design, there are basically two types of polygons: solid polygon without holes inside, and polygons with holes inside. Holes are always inside a hull polygon so we don't need functions to create holes. The polygons vertex order (counter-clockwise or clockwise) could be used to define they are hull or hole.

BTW, in klayout DB module, a polygon always have a hull poly and could have zero or more hole poly. So we don't need use the vertex order to classify hull or hole poly.

heitzmann, 2021-11-27:
I see, but neither GDSII nor OASIS file support polygons with holes. Gdstk doesn't modify the order of vertices, so you could interpret your layout in this manner, but there is no way to represent holes internally in the library. The only exceptions for this rule are offset, boolean and fracture functions, which will create polygons in the counter-clockwise direction regardless of the input orientation.

wylecho, 2021-11-27:
Got it. Thanks for your detailed explanation, @heitzmann . I will close this issue.

closed 2021-11-27 by wylecho