heitzmann/gdstk #72: Performance of boolean operation may be puzzle

Naluen, 2022-01-06:
I used gdspy a lot in the past, and really happy to see the new gdstk. Thank you for all the developers.
But the new performance of the boolean operation is really puzzle in gdstk.

For example, a polygon consist of a round and a rectangle just like a  ♂ without the arrow is really a normal contact type. In the gdspy, the boolean operation returns a real new polygon. So, it is easy to fillet the connection part of the round and the rectangle, forming a curve which could avoid the loss of high frequency RF signal. While, in gdstk, the boolean operation returns a list instead, and I can hardly find a way to get the same performance as gdspy.

 I really understand the change of boolean operation is for better performance of the programme, while i think the way of the gdspy should also be kept, maybe by another similar method.

Thank you.

heitzmann, 2022-01-06:
Hi @Naluen, the issue is code complexity. In gdspy, `boolean` returns a `PolygonSet`, not a `Polygon`, since any number of shapes can result from a boolean operation. The problem is that having to support both classes throughout the codebase increases complexity and worsens the overall performance.

To compensate for the loss of ergonomics, in gdstk, if I recall correctly, all functions that accept a polygon will also accept a sequence, so your code shouldn't change. Except fillet and fracture, which have to be performed per polygon:

```python
import gdstk

square = gdstk.rectangle((-10, -10), (10, 10))
circle = gdstk.ellipse((0, 0), 11)

contact = gdstk.boolean(square, circle, "not")
for poly in contact:
    poly.fillet(1)

gdstk.Cell("CONTACT").add(*contact).write_svg("contact.svg", 10)
```

Naluen, 2022-01-07:
Thank you so much for your reply. 

I really understand the complexity. Anyway, gdspy is good enough, so I can keep using it.

While I still want to explain why I proposed the issue. I just draw a ugly image to describe my issue. 
[image]

Just look at the connection part. In the left, there are sharp (OK, not so sharp) angles between rectangle and the circle, but in the right one, there are not. The kind of angles lead to signal loss if the frequency is high enough.

Thank you again.

heitzmann, 2022-01-07:
I see the difficulty you have. Filleting round shapes is a problem in both libraries because they don't have the concept of circles. A quick solution would be something like this:
```python
import numpy
import gdstk

def contact_pad(trace_width, pad_radius, fillet_radius, tolerance=0.01, layer=0, datatype=0):
    assert pad_radius > 0.5 * trace_width
    a = 0.5 * trace_width + fillet_radius
    c = pad_radius + fillet_radius
    b = (c ** 2 - a ** 2) ** 0.5
    alpha = numpy.arccos(a / c)

    curve = gdstk.Curve((b, -0.5 * trace_width), tolerance=tolerance)
    curve.arc(fillet_radius, 0.5 * numpy.pi, 0.5 * numpy.pi + alpha)
    curve.arc(pad_radius, 1.5 * numpy.pi + alpha, 0.5 * numpy.pi - alpha)
    curve.arc(fillet_radius, -0.5 * numpy.pi - alpha, -0.5 * numpy.pi)

    return gdstk.Polygon(curve.points(), layer=layer, datatype=datatype)

trace_width = 3

pad_cell = gdstk.Cell("PAD")
pad_cell.add(contact_pad(trace_width, 5, 2))

main_cell = gdstk.Cell("MAIN")

trace = gdstk.FlexPath([(0, 0), (25, 0), (25, 15)], trace_width, joins="round")

trace_spine = trace.spine()
pad1 = gdstk.Reference(pad_cell, trace_spine[0])
pad2 = gdstk.Reference(pad_cell, trace_spine[-1], -0.5 * numpy.pi)

main_cell.add(trace, pad1, pad2)

main_cell.write_svg("example.svg", 10)
```

But an easier to use solution would be using the `ends` argument in FlexPath:
```python
import numpy
import gdstk

def pad_function(pad_radius, fillet_radius, tolerance=0.01):
    def _f(p0, v0, p1, v1):
        p0 = numpy.array(p0)
        v0 = numpy.array(v0)
        p1 = numpy.array(p1)
        v1 = numpy.array(v1)

        half_trace_width = 0.5 * numpy.sqrt(numpy.sum((p0 - p1) ** 2))
        assert pad_radius > half_trace_width

        a = half_trace_width + fillet_radius
        c = pad_radius + fillet_radius
        b = (c ** 2 - a ** 2) ** 0.5
        alpha = numpy.arccos(a / c)
        gamma = numpy.arctan2(v0[1], v0[0])

        curve = gdstk.Curve(p0 - v0 * b, tolerance=tolerance)
        curve.arc(fillet_radius, gamma + 0.5 * numpy.pi, gamma + 0.5 * numpy.pi - alpha)
        curve.arc(
            pad_radius, gamma - 0.5 * numpy.pi - alpha, gamma + 0.5 * numpy.pi + alpha
        )
        curve.arc(fillet_radius, gamma - 0.5 * numpy.pi + alpha, gamma - 0.5 * numpy.pi)
        return curve.points()

    return _f

trace = gdstk.FlexPath(
    [(0, 0), (25, 0), (30, 15)], 3, joins="round", ends=pad_function(5, 2)
)

main_cell = gdstk.Cell("MAIN")
main_cell.add(trace)

main_cell.write_svg("example.svg", 10)
```

heitzmann, 2022-01-07:
FYI, I turned your use case in a [how-to](https://heitzmann.github.io/gdstk/how-tos.html#connection-pads) in the docs… it's a great example for custom ends.

Naluen, 2022-01-12:
Excellent！Thank you so much and sorry for late reply.

closed 2022-01-13 by heitzmann