heitzmann/gdstk #79: gds comparison fails after read-write.

nescirem, 2022-02-25:
When I used gdstk to read in the gds file containing the structure of the empty cell and the AREF referenced to the empty cell, and then wrote it out as gds, I found that the file structure had changed and the file size had become larger.

**code**:
```cpp
Library lib = read_gds("source.gds", 0, 1e-2, NULL, &error_code);
lib.write_gds("target.gds", 0, NULL);
```

**source.gds structure**:
```
AREF
SNAME "empty_CELL"
STRANS ( No_Refl )
ANGLE 90
COLROW 1 55
XY (
    202916 4283316
    202916 4283316
    179816 4283316
)
ENDEL
...
BGNSTR 2022/02/15 15:58:04 2022/02/15 15:58:04
STRNAME "empty_CELL"
ENDSTR
```

**target.gds structure**:
```
SREF
SNAME "empty_CELL"
STANS ( No_Refl )
ANGLE 90
MAG 1
XY (
    180236 4267076
)
ENDEL
SREF
SNAME "empty_CELL"
STANS ( No_Refl )
ANGLE 90
MAG 1
XY (
    180656 4267076
)
ENDEL
...  // 52 SREFs are omitted here
SREF
SNAME "empty_CELL"
STANS ( No_Refl )
ANGLE 90
MAG 1
XY (
    202916 4267076
)
ENDEL
...
BGNSTR 2022/02/25 15:58:04 2022/02/15 15:58:04
STRNAME "empty_CELL"
ENDSTR
```

nescirem, 2022-02-25:
gdstk seems to break up some 1xN AREF into SREF, I would like to know why and how to control this.

nescirem, 2022-02-25:
> gdstk seems to break up some 1xN AREF into SREF, I would like to know why and how to control this.

Is this preparation for the OASIS random distribution type Type 4-7?
[image]

heitzmann, 2022-02-25:
@nescirem The AREF should be stored as and AREF in the target GDSII. There was a problem detecting the repetition as regular when one of the displacement vectors was zero. Should be fixed now.

nescirem, 2022-02-28:
The problem still exists in the latest release. Is there anything I can do to help?

heitzmann, 2022-02-28:
Can you attach de gds file in which you see the problem? From the original description, I don't see the issue any longer.

nescirem, 2022-03-02:
I need to process the relevant files before sending them to you, maybe tomorrow.

nescirem, 2022-03-03:
The gds files have been sent to you via email, sorry the attached image in the email seems to have been sent unsuccessfully due to network problems, repost the attached images here: 
[image] 
fig.1 the source gds file which has 1x17 cell_16 (AREF)

[image]
fig.2 the gds file processed by gdstk that break cell_16 into SREFs

heitzmann, 2022-03-10:
@nescirem I believe the arrays are properly detected now. Let me know if it works with commit 5ffff20!

nescirem, 2022-03-11:
I have tested dozens of examples so far without finding similar problems.
we will test hundreds of examples next, and feed back the results here.

nescirem, 2022-03-17:
Hundreds of gds files have been tested and found everything is working fine, thank you very much for your quick response.:smile:

nescirem, 2022-04-15:
In the latest code, there are some AREF directions that will be changed
[image]

heitzmann, 2022-04-18:
@nescirem Can you send me the example (or the part of the layout) that gives the wrong output?

nescirem, 2022-04-19:
@heitzmann It may take a few days to modify and export to you. I will also be happy to help fix the problem and make a pull request.

heitzmann, 2022-04-27:
@nescirem Thanks again for the data and report. I've added a bunch of tests in the newest commit to make sure all case I could think of pass the array detection algorithm. Please, let me know if you find any remaining bugs.