heitzmann/gdstk #80: msvc++ prompts Compiler Error C2605

nescirem, 2022-02-28:
I‘m trying to build gdstk on Windows platform with msvc++, and found two pieces of code that will cause msvc++ compilation problem:

**src/cell.cpp:794**
```cpp
...
ErrorCode error_code = ErrorCode::NoError;
char* buffer = (char*)allocate(strlen(name) + 1);
// NOTE: Here be dragons if name is not ASCII.  The GDSII specification imposes ASCII-only
// for strings, but who knows…
char* d = buffer;
for (char* c = name; *c != 0; c++, d++) *d = *c == '#' ? '_' : *c;
*d = 0;
...
```
The above code will cause an "Undefined Reference" error, which can be solved by rewriting it as follows:

```cpp
...
ErrorCode error_code = ErrorCode::NoError;
char* d = NULL;
char* buffer = (char*)allocate(strlen(name) + 1);
// NOTE: Here be dragons if name is not ASCII.  The GDSII specification imposes ASCII-only
// for strings, but who knows…
d = buffer;
for (char* c = name; *c != 0; c++, d++) *d = *c == '#' ? '_' : *c;
*d = 0;
...
```

The same problem exists in **src/reference.cpp:515**.

heitzmann, 2022-02-28:
Can you paste the full error report? [C2605](https://docs.microsoft.com/en-us/cpp/error-messages/compiler-errors-2/compiler-error-c2605?view=msvc-170) doesn't seem to be 'undefined reference'.

nescirem, 2022-03-01:
:woozy_face: Sorry, it's actually `error C2065: 'd' : undeclared identifier`

heitzmann, 2022-03-01:
That still doesn't provide the full error report. It would also be good to know which compiler version you're using. The piece of code you highlighted seems to [compile just fine with MSVC](https://godbolt.org/#g:!((g:!((g:!((h:codeEditor,i:(filename:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,selection:(endColumn:2,endLineNumber:41,positionColumn:2,positionLineNumber:41,selectionStartColumn:2,selectionStartLineNumber:41,startColumn:2,startLineNumber:41),source:'%23include+%3Cstdio.h%3E%0A%23include+%3Cstdlib.h%3E%0A%23include+%3Cstdint.h%3E%0A%23include+%3Cstring.h%3E%0A%0Astruct+Polygon+%7B%0A%7D%3B%0A%0Atypedef+bool+(*PolygonComparisonFunction)(Polygon*+const%26,+Polygon*+const%26)%3B%0A%0Astruct+Cell+%7B%0A++++char*+name%3B%0A%0A++++int+to_svg(FILE*+out,+double+scaling,+uint32_t+precision,+const+char*+attributes,%0A+++++++++++++++++++++++PolygonComparisonFunction+comparison)+const%3B%0A%7D%3B%0A%0Aint+Cell::to_svg(FILE*+out,+double+scaling,+uint32_t+precision,+const+char*+attributes,%0A+++++++++++++++++++++++PolygonComparisonFunction+comparison)+const+%7B%0A++++int+error_code+%3D+0%3B%0A++++char*+buffer+%3D+(char*)malloc(strlen(name)+%2B+1)%3B%0A++++//+NOTE:+Here+be+dragons+if+name+is+not+ASCII.++The+GDSII+specification+imposes+ASCII-only%0A++++//+for+strings,+but+who+knows%E2%80%A6%0A++++char*+d+%3D+buffer%3B%0A++++for+(char*+c+%3D+name%3B+*c+!!%3D+0%3B+c%2B%2B,+d%2B%2B)+*d+%3D+*c+%3D%3D+!'%23!'+%3F+!'_!'+:+*c%3B%0A++++*d+%3D+0%3B%0A%0A++++if+(attributes)+%7B%0A++++++++fprintf(out,+%22%3Cg+id%3D%5C%22%25s%5C%22+%25s%3E%5Cn%22,+buffer,+attributes)%3B%0A++++%7D+else+%7B%0A++++++++fprintf(out,+%22%3Cg+id%3D%5C%22%25s%5C%22%3E%5Cn%22,+buffer)%3B%0A++++%7D%0A++++return+0%3B%0A%7D%0A%0Aint+main(int+argc,+char*+argv%5B%5D)+%7B%0A++++Cell+c+%3D+%7B0%7D%3B%0A++++c.name+%3D+%22NAME%22%3B%0A++++c.to_svg(stdout,+1.0,+1,+NULL,+NULL)%3B%0A++++return+0%3B%0A%7D'),l:'5',n:'0',o:'C%2B%2B+source+%231',t:'0')),k:56.096284667713235,l:'4',n:'0',o:'',s:0,t:'0'),(g:!((g:!((h:compiler,i:(compiler:vcpp_v19_29_VS16_11_x64,filters:(b:'0',binary:'1',commentOnly:'0',demangle:'0',directives:'0',execute:'0',intel:'0',libraryCode:'0',trim:'1'),flagsViewOpen:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,libs:!(),options:'',selection:(endColumn:1,endLineNumber:1,positionColumn:1,positionLineNumber:1,selectionStartColumn:1,selectionStartLineNumber:1,startColumn:1,startLineNumber:1),source:1,tree:'1'),l:'5',n:'0',o:'x64+msvc+v19.29+VS16.11+(C%2B%2B,+Editor+%231,+Compiler+%231)',t:'0')),k:43.903715332286765,l:'4',m:50,n:'0',o:'',s:0,t:'0'),(g:!((h:output,i:(compiler:1,editor:1,fontScale:14,fontUsePx:'0',tree:'1',wrap:'1'),l:'5',n:'0',o:'Output+of+x64+msvc+v19.29+VS16.11+(Compiler+%231)',t:'0')),header:(),l:'4',m:50,n:'0',o:'',s:0,t:'0')),k:43.903715332286765,l:'3',n:'0',o:'',t:'0')),l:'2',n:'0',o:'',t:'0')),version:4)

We have also a [conda forge distribution](https://github.com/conda-forge/gdstk-feedstock) that has been working on Windows for quite some time now.

nescirem, 2022-03-03:
```
Severity    Code    Description Project Path    File    Line Suppression State
Error   C2065   'd': undeclared identifier  gdstk   D:\Test\gtest-master\src    D:\Test\gtest-master\src\cell.cpp   794
Error   C2065   'd': undeclared identifier  gdstk   D:\Test\gtest-master\src    D:\Test\gtest-master\src\cell.cpp   795
Error   C2065   'd': undeclared identifier  gdstk   D:\Test\gtest-master\src    D:\Test\gtest-master\src\reference.cpp   515
Error   C2065   'd': undeclared identifier  gdstk   D:\Test\gtest-master\src    D:\Test\gtest-master\src\reference.cpp   516
```
Microsoft Visual Studio 2017 Version 15.9.24
Windows SDK Version: 10.0.17763.0

nescirem, 2022-03-04:
That works, thank you very much for your patience! :smile:

closed 2022-03-04 by nescirem