heitzmann/gdstk #81: Question: why are `layers` and `types` treated independently in `Cell.filter`?

ottumm, 2022-03-02:
I was using `Cell.filter` and noticed this in the documentation:

> An element will be removed if the following check returns True: (element.layer in layers) op (element.datatype in types)

This implies that the `layers` list and the `datatype` list are treated independently of each other, rather than as `layer, datatype` pairs. The implementation is consistent with the documentation here: https://github.com/heitzmann/gdstk/blob/574ff22de862b289c506358467db4f1c1e1aebe1/python/cell_object.cpp#L856

Given that a single logical layer is represented by a `layer, datatype` pair of ints, this design is surprising, because passing more than one value into `layer` or `datatype` would filter out more layers than I would expect.

E.g. if I have three layers `(0, 0)`, `(1, 0)`, `(0, 1)` and I want to filter out the last two, I need to make two calls: `filter(layers=[1], types=[0], "and")`, `filter(layers=[0], types=[1], "and")`. Because if I make a single call like `filter(layers=[1, 0], types=[0, 1], "and")` it will also filter out the first layer `(0, 0)`.

So, two questions:

1. Is there a use case for the existing behavior?
2. Would you be open to adding a parameter that is a list of `layer, datatype` tuples (for `filter` and every other method that deals with layers), so that exact behavior can be specified?

ottumm, 2022-03-02:
I just saw that `read_gds` has the behavior I'm requesting: it takes in an iterable of tuples, each of which represents a unique `layer, datatype` pair.

heitzmann, 2022-03-02:
The way it is currently implemented gives more freedom for tailoring the filter. For instance, in your example, you could use `filter(layers=[1], types=[1], 'or')` or `filter(layers=[0], types=[0], 'nand')` to get the desired filtering. More complex examples might not be possible to solve in a single call, but this signature is definitely more flexible than the one in `read_gds`.
 
However, I understand the use case you mention, which is implemented in `reag_gds`, and I'm not particularly happy with the inconsistency among those two functions, so we may decide to change `Cell.filter` if there is a strong reason to do so.

ottumm, 2022-03-02:
The flexibility makes sense, but only if I know all the possible layer/datatype pairs at the time that I’m implementing the query, which isn’t feasible in my case (there are hundreds, and the list will change in the future). Since I don’t know that full set, it is impossible to write a correct query given then current interface, unless I limit myself to one layer and one datatype per call to filter, which is what I’m currently doing. 

I would strongly support matching this interface to read_gds.

ottumm, 2022-03-03:
Excellent, thank you so much!

closed 2022-03-03 by heitzmann