heitzmann/gdstk #84: Read Large OAS file

XDean, 2022-03-10:
I have `.oas` file that larger than 5GB with billions polygons.

When I use `gdstk` to read, it used minutes time and memory > 5GB.

Is there a way to handle the large oas without read it all into memory at first time? Something like `numpy.memmap`.

Or is there a way to read it as several parts?

heitzmann, 2022-03-10:
@XDean at this point, there is no way to read it partially. In the future, I plan to add a filter functionality similar to read_gds, but there are other issues I'm working on before that.

MarkMa1990, 2022-07-09:
I think there is always memory problem while reading large oas files, it consumes very big memory resources suddenly.