heitzmann/gdstk #88: Should a warning be given when some record does not satisfy the tolerance?

nescirem, 2022-03-17:
I found that gdstk will silently rounding some record that does not meet the tolerance I setted, should we give a warning after a rounding is happend?

The structure shown below is changed because it does not satisfy the condition of tolerance=1e-2:
[image][image]

heitzmann, 2022-03-17:
@nescirem There are several places where a tolerance must be used. What function are you using that you suggest including the warning? Or, how did you process the file, just opening and saving, did you do any operations?

nescirem, 2022-03-18:
Maybe any layout changes not directly caused by the user should provide a warning?🤔 

At this stage, I am not doing anything with the layout other than reading and writing layout files with gdstk. Next I will try to add a graphical interface where the user can view and edit the layout. Next I will try to add a graphical interface where the user can view and edit the layout. By then, it is acceptable for the software to perform a silent rounding operation when the user creates a new record. However, if rounding occurs silently when exporting the layout file, the user will find that the layout is not what he saw when he last saved, which may confuse users and is what I want to avoid.

heitzmann, 2022-03-18:
That would basically raise a warning for all layouts. Rounding is unavoidable in GDSII and OASIS and, by using floating point coordinates internally, there are always going to be small changes when writing a file. Maybe the worse issue is when 2 coordinates are rounded to the same point and automatically joined. That's necessary for other reasons (such as numerical reliability when calculating a path direction). To avoid surprises, the default values for tolerances (1e-2) is 10 times larger than the default GDSII grid (precision / unit = 1e-3), so that shapes created from the default parameters are not significantly affected.

The issue you're running into is probably because the default tolerance for loading a file is 1e-2, but that's for historical reasons only. If you try to change the default tolerance when reading the file from 1e-2 to 1e-3 (or, even better, to the file grid size), you probably won't see this issue any longer.

That said, it looks like the current default is not appropriate, so changing it to 1e-3 (to correspond to match the default writing grid) or to the file grid size would be safer. In any case, the user must be aware of rounding issues in both file formats. There's a whole section on the documentation only about this issue.