heitzmann/gdstk #89: Newbie question incoming : "gdstk.h No such file or directory"

mark-fein, 2022-04-21:
Hello, I am pretty new to C++ overall and am having some trouble generating the initial red rectangle in gdstk. I have copy-pasted source code below into a "Hello World" default Project in Eclipse but am getting the above error. I assume its because I need to "point" or include the referenced header file or somewhere include it in list of things to look [in binaries for (much like <iostream> and <stdio.h> live)].

See below for reference code and screen shot for project explorer window:

  //============================================================================
// Name        : This_is_the_test.cpp
// Author      : Mark F
// Version     :
// Copyright   : Your copyright notice
// Description : Hello World in C++, Ansi-style
//============================================================================

#include <iostream>

int main() {
#include <stdio.h>

#include "gdstk.h"

using namespace gdstk;

int main(int argc, char* argv[]) {
    Library lib = {.unit = 1e-6, .precision = 1e-9};
    lib.name = copy_string("library", NULL);

    Cell cell = {0};
    cell.name = copy_string("FIRST", NULL);
    lib.cell_array.append(&cell);

    Polygon rect = rectangle(Vec2{0, 0}, Vec2{2, 1}, make_tag(0, 0));
    cell.polygon_array.append(&rect);

    lib.write_gds("first.gds", 0, NULL);
    lib.write_oas("first.oas", 0, 6, OASIS_CONFIG_DETECT_ALL);
    cell.write_svg("first.svg", 10, 6, NULL, NULL, "#222222", 5, true, NULL);

    rect.clear();
    cell.clear();
    lib.clear();
    return 0;
}
}

[image]

mark-fein, 2022-04-22:
Associated SE Thread over here as well [Linky](https://stackoverflow.com/questions/71958229/newbie-question-incoming-gdstk-h-no-such-file-or-directory)

heitzmann, 2022-04-25:
@mark-fein You can use CMake to build and install gdstk as a shared/dynamic library and configure your eclipse project to link to the installed version of gdstk and find the installed header (gdstk.h).

mark-fein, 2022-04-25:
I have successfully installed/built it here: C:\msys64\home\me\projects\gdstk with following files. Now do I run from Anaconda prompt or within a C++ IDE?

[image]

Thanks for all the help so far

mark-fein, 2022-04-25:
> Tried that (as alluded to in post) however I now get the following error:
> 
> ```
> markfein124@l-ul71kght6q MINGW64 ~/projects/gdstk
> $ cmake -S . -B build -G"MSYS Makefiles"
> CMake Error: Could not create named generator MSYS Makefiles
> 
> Generators
> * Unix Makefiles               = Generates standard UNIX makefiles.
>   Ninja                        = Generates build.ninja files.
>   Ninja Multi-Config           = Generates build-<Config>.ninja files.
>   CodeBlocks - Ninja           = Generates CodeBlocks project files.
>   CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
>   CodeLite - Ninja             = Generates CodeLite project files.
>   CodeLite - Unix Makefiles    = Generates CodeLite project files.
>   Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
>   Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.
>   Kate - Ninja                 = Generates Kate project files.
>   Kate - Unix Makefiles        = Generates Kate project files.
>   Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
>   Sublime Text 2 - Unix Makefiles
>                                = Generates Sublime Text 2 project files.
> ```
> 
> Thoughts?

I fixed this. See comment below

heitzmann, 2022-04-25:
How did you compile? Do you have a log of the compilation messages? Compiling on windows is troublesome. If you used Anaconda for compilation, you should have the python interface working. On the Anaconda prompt you can try `python -c "import gdstk"` to see if any errors show up.

To use with C++ you'll have to compile with cmake and check where it installs header files and the resulting lib on Windows. In your IDE you'll have to configure your project to find those files.

mark-fein, 2022-04-25:
These are the header files, correct?

[image]
These are in the a copy-able path. How to add this to a project in IDE (say eclipse) or another (if you have one you prefer)?

mark-fein, 2022-04-25:
> How did you compile? Do you have a log of the compilation messages? Compiling on windows is troublesome. If you used Anaconda for compilation, you should have the python interface working. On the [...]

Tried in Anaconda prompt and showed no error as far as I can tell

[image]

heitzmann, 2022-04-25:
I don't use an IDE—I can't help you there, sorry. On the command line it's just a flag.

Your python interface seems to be working correctly.

mark-fein, 2022-04-25:
Is there any preference to running in python vs c++?

heitzmann, 2022-06-07:
@mark-fein Sorry for taking so long to reply. I missed your last comment.
In C++ you gain in efficiency. For large layouts, it can be a huge advantage. However, you'll be responsible for memory management, which adds complexity. If you're comfortable with C, it shouldn't be a problem.
The advantage of using Python is the easy of quickly prototyping and iterating over different layouts.

closed 2022-10-28 by heitzmann