heitzmann/gdstk #90: Cmake problem for windows compilation

paulgoulain, 2022-04-28:
Hi,

I managed to compile gdstk and install it on windows using Python 3.10 and MSVC 2022 (14.31). However to succeed, I had to make a change to the CMakeList.txt file.

During the compilation, I had errors because the CMake PRIVATE_LIBS variable has the value:
optimized C:/vcpkg/installed/x64-windows/lib/zlib.lib debug C:/vcpkg/installed/x64-windows/debug/lib/zlibd.lib

But the optimized and debug words were making the compilation failing. I had to make the following changes to CMakeLists file:

```cmake
foreach(IT ${ZLIB_LIBRARIES})
    if(IT STREQUAL "optimized")
        continue()
    elseif(IT STREQUAL "debug")
        continue()
    else()
        set(PRIVATE_LIBS "${PRIVATE_LIBS} ${IT}")
    endif()
endforeach()
```

Basically removing the "optimized" and "debug" words from the string.
Does this change is problematic for a Linux compilation ? If it isn't, I can make a PR to allow easier Windows compilation.

Cheers,
Paul

heitzmann, 2022-04-28:
@paulgoulain We have many Windows users and our CI running the installation on Windows without issues. It seems to me that the problem you're seeing is related to your installation of zlib, not gdstk, because I don't think those arguments should be present in ZLIB_LIBRARIES.

paulgoulain, 2022-04-28:
I'm using vcpkg which provides the compiled zlib. I can't find a link to your CI, where is it ?

heitzmann, 2022-04-28:
It's on https://github.com/heitzmann/gdstk/actions

paulgoulain, 2022-04-28:
After digging a little in the CI, I noticed the Windows' builds are all made by a conda environment. Do you have a CI for a regular Python install on Windows ?

heitzmann, 2022-04-29:
No, I've tried to compile it locally at some point, but there was a dependency on Lapack that would not work, so I've ended up usign conda for it. The dependency has been dropped, but I haven't tried to go back to a "native" Windows build.

closed 2022-09-07 by heitzmann