heitzmann/gdstk #91: import crash

frankxxu, 2022-05-04:
Python 3.7.4 (default, Aug 13 2019, 20:35:49)
[GCC 7.3.0] :: Anaconda, Inc. on linux
>>>imoport gdstk
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
ImportError: /home/xxx/lib/python3.7/site-packages/gdstk-0.8.2-py3.7-linux-x86_64.egg/gdstk.cpython-37m-x86_64-linux-gnu.so: undefined symbol: __ZSt28__throw_bad_array_new_lengthv

gcc version 11.2.1 20210728 (Red Hat 11.2.1-1) (GCC)
Does anyone has idea?

frankxxu, 2022-05-04:
to be clear:
1)  python 3.7.4 in centos7. Used gcc 7.3.0.
2) I used gcc 11.2.1 to compiled zlib.
3) cmake3 3.17.5

heitzmann, 2022-05-04:
@frankxxu This seems to be a C++ library version problem. Please take a look at https://github.com/stan-dev/pystan/issues/294 (looks related to your issue).