heitzmann/gdstk #93: A mistake in gdstk document; about how to initialize objects in C++

HecatePhy, 2022-05-11:
Taking the [first layout](https://heitzmann.github.io/gdstk/gettingstarted.html#first-layout) in C++ as example:
`
Library lib = {.unit = 1e-6, .precision = 1e-9};
`
This initialization can cause a `non-trivial designated initializers not supported` error with some compilers.
The initialization is supposed to be compatible with the struct definition:
`
Library lib = {.name = NULL, .unit = 1e-6, .precision = 1e-9};
`
The same problem occurs in the document many times, such as [Polygons](https://heitzmann.github.io/gdstk/gettingstarted.html#polygons):
`
poly->point_array.extend({.count = COUNT(points), .items = points});
`
Note that `extend` function takes an `Array` object which requires `{.capacity = , .count = , .items = }`.
It is kind of the authors of the document to go through all the C++ examples in the document and fix this problem.

HecatePhy, 2022-05-12:
> That's all g++'s fault for being more tolerant! Just kidding! I'll try to go over all such occurrences when I have some time, thanks for reporting!

🤩

heitzmann, 2022-06-07:
I'll close this for now, as I could not find any other instances with these errors. Please @HecatePhy, let me know if you find any others.

closed 2022-06-07 by heitzmann